tdorssers/TeslaPy #1: HTTP Error during Authenticate

kdvlr, 2019-09-21:
Just found this repo. Seems very well put together. I tried using the cli.py as well as importing the teslapy in my own python script. It fails with the following error. 

```
Traceback (most recent call last):
  File "./test.py", line 3, in <module>
    tesla.fetch_token()
  File "/docker/homeassistant/config/python_scripts/TeslaPy/teslapy.py", line 69, in fetch_token
    response = self.api('AUTHENTICATE', data=data)
  File "/docker/homeassistant/config/python_scripts/TeslaPy/teslapy.py", line 162, in api
    return self.request(endpoint['TYPE'], uri, data=kwargs)
  File "/docker/homeassistant/config/python_scripts/TeslaPy/teslapy.py", line 55, in request
    response.raise_for_status()  # Raise HTTPError, if one occurred
  File "/usr/lib/python3/dist-packages/requests/models.py", line 940, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 401 Client Error: invalid_request. The request is missing a required parameter, includes an unsupported parameter value, or is otherwise malformed for url: https://owner-api.teslamotors.com/oauth/token
```
I tried to recreate the cache.json with a valid token, but that did not seem to work either.

kdvlr, 2019-09-21:
I solved it by changing data=data to **data.

closed 2019-09-21 by kdvlr

tdorssers, 2019-09-22:
Thanks! I have fixed fetch_token and refresh_token as well.