tdorssers/TeslaPy #5: 444 Client Error: Custom for URL

jhattonc, 2021-01-29:
I have been using this module for a while with no issues until recently, I have made no changes to previously working code and now I am getting  a 444 Client Error: CUSTOM for url: https://owner-api.teslamotors.com/oauth/token when calling fetch_token()  at first I thought that it was an issue with the tesla server, but the tesla app and another third party app I use are both connecting fine.   any ideas what I can try to get a token?

jhattonc, 2021-01-29:
my other apps have now started to drop their connections as tokens have expired so I believe it is on Tesla's side not the script

jhattonc, 2021-01-29:
I was wrong... it looks like the  /oauth/token endpoint is returning a Endpoint_Depreciated 400 error

tdorssers, 2021-02-01:
It appears Tesla no longer supports the old login method. I will look into this.

tdorssers, 2021-02-06:
I have updated the script and Tesla's new SSO and MFA are now supported

jhattonc, 2021-02-06:
I have downloaded the new code and am now getting 

    from requests_oauthlib import OAuth2Session
ModuleNotFoundError: No module named 'requests_oauthlib

I removed and reloaded requests_oauthlib but it is still giving me this message...  is there a newer version i need from somewhere

Just using the sudo apt-get install python-requests-oauthlib to reload it

tdorssers, 2021-02-07:
Please try 'sudo apt-get install python3-requests-oauthlib'.

jhattonc, 2021-02-07:
Thanks, that did it, I am back up and running...

closed 2021-02-07 by jhattonc

tdorssers, 2021-02-14:
I have also added support for multiple authentication devices and token refreshing.