tdorssers/TeslaPy #6: Issues running on Raspberry pi

erbmur, 2021-02-01:
Hello,

I have a simple script that runs fine on my PC, but when I copy it across to my raspberry pi, I get the following error:

`  File "TeslaDisplay.py", line 19, in <module>
    tesla.fetch_token()
  File "/home/pi/Scripts/TeslaDisplay/teslapy/teslapy.py", line 71, in fetch_token
    email=self.email, password=self.password)
  File "/home/pi/Scripts/TeslaDisplay/teslapy/teslapy.py", line 165, in api
    return self.request(endpoint['TYPE'], uri, data=kwargs)
  File "/home/pi/Scripts/TeslaDisplay/teslapy/teslapy.py", line 58, in request
    response.raise_for_status()  # Raise HTTPError, if one occurred
  File "/home/pi/.local/lib/python3.7/site-packages/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: endpoint_deprecated:_please_update_your_app for url: https://owner-api.teslamotors.com/oauth/token
`

Any ideas what I might be missing?

closed 2021-02-01 by erbmur

tdorssers, 2021-02-06:
I have updated the script and Tesla's new SSO and MFA are now supported

erbmur, 2021-02-06:
You will have to forgive me but I have been trying to teach myself python for only a short while.  I had managed to work backwords from your previous build to get my project working, but with this new update I don't seem to be able to work out how to use your API.

Previously I would use the below in the own code:

import teslapy

with teslapy.Tesla(EMAIL, PASSWORD, CLIENT_ID, CLIENT_SECRET) as tesla:
    tesla.fetch_token()
    vehicles = tesla.vehicle_list()
    Azurite = vehicles[0]
    Azurite_State = Azurite['state']
    if Azurite_State != 'asleep':
        Azurite_Raw = Azurite.get_vehicle_data()
        NearestCharges = Azurite.get_nearby_charging_sites()
    else:
        with open(os.path.join(__location__,'Azurite_SavedData.json')) as jsonfile:
            Azurite_Raw = json.loads(jsonfile.read())
            jsonfile.close

but with the new update I see there are some different arguments that I can't seem to work out.  If you could point me in the right direction that would be great?

tdorssers, 2021-02-07:
Please change the line 'with teslapy.Tesla(EMAIL, PASSWORD, CLIENT_ID, CLIENT_SECRET) as tesla:' to 'with teslapy.Tesla(EMAIL, PASSWORD) as tesla:'. In case you have enabled MFA, then you need to specify a method as the 3rd argument. This method should return the passcode.

erbmur, 2021-02-09:
Thank you so much.  I really appreciate the effort you have put into this project and taking the time to answer my question.

tdorssers, 2021-02-14:
No problem. I have also added support for multiple authentication devices and token refreshing.