tdorssers/TeslaPy #11: Image renderer not working

mikemoerk, 2021-02-27:
compose_image seems to be passing invalid query params to the Tesla image compositor.

Here is the URL being accessed for my vehicle:

https://static-assets.tesla.com/v1/compositor/?model=ms&bkba_opt=1&view=STUD_3QTR&size=640&options=AD15,MDL3,PBSB,RENA,BT37,ID3W,RF3G,S3PB,DRLH,DV2W,W39B,APF0,COUS,BC3B,CH07,PC30,FC3P,FG31,GLFR,HL31,HM31,IL31,LTPB,MR31,FM3B,RS3H,SA3P,STCP,SC04,SU3C,T3CA,TW00,TM00,UT3P,WR00,AU3P,APH3,AF00,ZCST,MI00,CDM0

Note the image is not right.

Here is the URL that tesla.com uses for my vehicle (which renders accurately):

https://static-assets.tesla.com/v1/compositor/?model=ms&view=STUD_3QTR_V2&size=1440&options=MDLS,MTS07,PPSB,RFFR,WTMS,INB3C,PI01,APBS,SC04,CPF1&bkba_opt=1&context=design_studio_2

mikemoerk, 2021-02-27:
Did a little investigation.  When you go to tesla.com and look at your vehicle, it gets a different list of options by calling

https://www.tesla.com/teslaaccount/ownership-api/vehicle-details?rn=RN#########&locale=en_US

That gives me the following specs:
```
[
  {
    "code": "$MTS07",
    "name": "Model S Long Range Plus",
    "long_name": "Long Range Plus All-Wheel Drive"
  },
  {
    "code": "$PPSB",
    "name": "Deep Blue Metallic",
    "long_name": "Deep Blue Metallic"
  },
  {
    "code": "$WTMS",
    "name": "19'' Tempest Wheels",
    "long_name": "19'' Tempest Wheels"
  },
  {
    "code": "$INB3C",
    "name": "Cream Premium Interior with Oak Wood Décor",
    "long_name": "Cream Premium Interior with Oak Wood Décor"
  },
  {
    "code": "$APBS",
    "name": "Autopilot",
    "long_name": "Autopilot"
  },
  {
    "id": "SC04",
    "code": "$SC04",
    "name": "Pay Per Use Supercharging",
    "long_name": "Pay-as-you-go Supercharging"
  }
]
```
If I take those options, and plug them into the compositor, then I get a good image.

Haven't yet figured out how to get it to call the vehicle-details endpoint, since it has a different root URL.  Also not sure if the auth token will work with that endpoint.

tdorssers, 2021-02-28:
This is a known issue, the option codes returned by the owner API are not valid anymore.
The Dutch tesla.com site doesn't show my vehicle. And it doesn't give me the RN, so I can't reproduce this.
You can try to use the access_token of sso_token.

closed 2021-03-26 by tdorssers