tdorssers/TeslaPy #12: Help

gkpw2, 2021-03-11:
Hi,

I'm interested in using the telsapy for accessing more information about the Tesla Powerwall battery but having a bit of trouble.

I'm new to Python, and api, so it's all a bit of jumble for me at the moment.

So, I've download Python (on Windows) and did the following
python -m pip install teslapy
python -m pip install requests_oauthlib geopy

No problem with doing that, all successful.

I've copied the cli.py file (actually I copied the text into Notepad and saved it as cli.py in the c:\python directory.

But when I execute the following command (where myemailaddress and mypassword is what I used to get access to the Tesla app on Android)

python cli.py -e myemailaddress -p mypassword -b

I get the following reply:-

Traceback (most recent call last):
  File "C:\Python\cli.py", line 90, in <module>
    main()
  File "C:\Python\cli.py", line 59, in main
    selected = prod =  tesla.vehicle_list() + tesla.battery_list()
AttributeError: 'Tesla' object has no attribute 'battery_list'

Is there something I'm supposed to put in the cli.py? What is wrong with line 59 and 90?

So, I don't know where the problem is or what I should be expecting (hoping it was something like the Powerwall data section posted)

Be much appreciated if someone can point me with troubleshooting as this is my first attempt at Python programming.

Thanks,
Graham

tdorssers, 2021-03-11:
The packaged version on PIP does not include Powerwall support yet. Please clone this repo to use Powerwall support. I will update the PIP package on a short term.

tdorssers, 2021-03-11:
Please run `python -m pip install --upgrade teslapy` to install the updated version and let me know the result.

closed 2021-03-26 by tdorssers