tdorssers/TeslaPy #15: list index out of range

aishlai, 2021-04-08:
2021-04-08 20:53:56,123 - teslapy - WARNING - compose_image requires options for the image to be accurate
Exception in Tkinter callback
Traceback (most recent call last):
  File "/Users/arvin/opt/anaconda3/lib/python3.8/tkinter/__init__.py", line 1883, in __call__
    return self.func(*args)
  File "/Users/arvin/opt/anaconda3/lib/python3.8/tkinter/__init__.py", line 804, in callit
    func(*args)
  File "gui.py", line 644, in process_update_dashboard
    self.dashboard.update_widgets()
  File "gui.py", line 316, in update_widgets
    self.ft.text(door[ve['ft']])
IndexError: list index out of range

tdorssers, 2021-04-08:
Can you share the output of the "vehicle_state" section from "py cli.py -e <YOUREMAIL> -w -g" please?

aishlai, 2021-04-09:
> Can you share the output of the "vehicle_state" section from `py cli.py -e <YOUREMAIL> -w -g` please?

I hidden some id info for you，use ******* to change

{
    "id": ****************,
    "vehicle_id": ****************,
    "vin": "****************",
    "display_name": "\u7279\u7279",
    "option_codes": "AD15,MDL3,PBSB,RENA,BT37,ID3W,RF3G,S3PB,DRLH,DV2W,W39B,APF0,COUS,BC3B,CH07,PC30,FC3P,FG31,GLFR,HL31,HM31,IL31,LTPB,MR31,FM3B,RS3H,SA3P,STCP,SC04,SU3C,T3CA,TW00,TM00,UT3P,WR00,AU3P,APH3,AF00,ZCST,MI00,CDM0",
    "color": null,
    "access_type": "OWNER",
    "tokens": [
        "****************",
        "****************"
    ],
    "state": "online",
    "in_service": false,
    "id_s": "****************",
    "calendar_enabled": true,
    "api_version": 13,
    "backseat_token": null,
    "backseat_token_updated_at": null,
    "command_signing": "off",
    "user_id": ****************,
    "charge_state": {
        "battery_heater_on": false,
        "battery_level": 39,
        "battery_range": 121.3,
        "charge_current_request": 16,
        "charge_current_request_max": 16,
        "charge_enable_request": true,
        "charge_energy_added": 44.59,
        "charge_limit_soc": 90,
        "charge_limit_soc_max": 100,
        "charge_limit_soc_min": 50,
        "charge_limit_soc_std": 90,
        "charge_miles_added_ideal": 186.5,
        "charge_miles_added_rated": 186.5,
        "charge_port_cold_weather_mode": false,
        "charge_port_door_open": false,
        "charge_port_latch": "Engaged",
        "charge_rate": 0.0,
        "charge_to_max_range": false,
        "charger_actual_current": 0,
        "charger_phases": null,
        "charger_pilot_current": 16,
        "charger_power": 0,
        "charger_voltage": 2,
        "charging_state": "Disconnected",
        "conn_charge_cable": "<invalid>",
        "est_battery_range": 155.59,
        "fast_charger_brand": "<invalid>",
        "fast_charger_present": false,
        "fast_charger_type": "<invalid>",
        "ideal_battery_range": 121.3,
        "managed_charging_active": false,
        "managed_charging_start_time": null,
        "managed_charging_user_canceled": false,
        "max_range_charge_counter": 0,
        "minutes_to_full_charge": 0,
        "not_enough_power_to_heat": null,
        "scheduled_charging_pending": false,
        "scheduled_charging_start_time": null,
        "scheduled_departure_time": 1617930000,
        "time_to_full_charge": 0.0,
        "timestamp": 1617949576261,
        "trip_charging": false,
        "usable_battery_level": 39,
        "user_charge_enable_request": null
[... 113 lines cut ...]
        "rt": 0,
        "sentry_mode": true,
        "sentry_mode_available": true,
        "smart_summon_available": true,
        "software_update": {
            "download_perc": 0,
            "expected_duration_sec": 2700,
            "install_perc": 1,
            "status": "",
            "version": " "
        },
        "speed_limit_mode": {
            "active": true,
            "current_limit_mph": 86.864114,
            "max_limit_mph": 90,
            "min_limit_mph": 50,
            "pin_code_set": true
        },
        "summon_standby_mode_enabled": false,
        "timestamp": 1617949576261,
        "valet_mode": false,
        "valet_pin_needed": true,
        "vehicle_name": "\u7279\u7279"
    }
}

aishlai, 2021-04-09:
> Can you share the output of the "vehicle_state" section from `py cli.py -e <YOUREMAIL> -w -g` please?

"vehicle_state": {
        "api_version": 13,
        "autopark_state_v2": "ready",
        "autopark_style": "dead_man",
        "calendar_supported": true,
        "car_version": "2020.47.104 f0c38cc0a86d",
        "center_display_state": 0,
        "df": 0,
        "dr": 0,
        "fd_window": 0,
        "fp_window": 0,
        "ft": 0,
        "is_user_present": false,
        "last_autopark_error": "no_error",
        "locked": true,
        "media_state": {
            "remote_control_enabled": true
        },
        "notifications_supported": true,
        "odometer": 621.833522,
        "parsed_calendar_supported": true,
        "pf": 0,
        "pr": 0,
        "rd_window": 0,
        "remote_start": false,
        "remote_start_enabled": true,
        "remote_start_supported": true,
        "rp_window": 0,
        "rt": 0,
        "sentry_mode": true,
        "sentry_mode_available": true,
        "smart_summon_available": true,
        "software_update": {
            "download_perc": 0,
            "expected_duration_sec": 2700,
            "install_perc": 1,
            "status": "",
            "version": " "
        },
        "speed_limit_mode": {
            "active": true,
            "current_limit_mph": 86.864114,
            "max_limit_mph": 90,
            "min_limit_mph": 50,
            "pin_code_set": true
        },
        "summon_standby_mode_enabled": false,
        "timestamp": 1617949576261,
        "valet_mode": false,
        "valet_pin_needed": true,
        "vehicle_name": "\u7279\u7279"
    }

tdorssers, 2021-04-09:
Please try the fix integrated in commit d9db5a0

aishlai, 2021-04-11:
> Please try the fix integrated in commit [d9db5a0](https://github.com/tdorssers/TeslaPy/commit/d9db5a064cc6cc5c07b2f03fa41183d5ec88c514)

It's worked，thank u

closed 2021-04-12 by tdorssers