tdorssers/TeslaPy #19: Unauthorized Access

stevieb9, 2021-05-07:
Two days ago, I noticed one of my devices which uses your software was failing to retrieve data from Tesla about my vehicle. I checked my official Tesla phone app, and for the first time since installing it, was requesting I re-enter my username and password.

I dug a little deeper, and both my custom script which uses the library, as well as the included `gui.py` script fail with the following error:

```
401 Client Error: Unauthorized for url: https://owner-api.teslamotors.com/api/1/vehicles
```

Is anyone else noticing such problems?

tdorssers, 2021-05-07:
Are you using the latest version? I just tried re-authentication and it works for me. Are you using two factor authentication?

stevieb9, 2021-05-11:
Updating to the most recent version corrected the issue. I apologize for the noise.