tdorssers/TeslaPy #20: Credentials rejected

MarlinMr, 2021-05-30:
I seem to be getting `ValueError: Credentials rejected`. It might be due to Tesla adding a captcha lately.

sillyfrog, 2021-06-01:
I'm having the same issue. No 2FA enabled on my account. Fresh install in a new venv, Python v3.9.1.

Using `menu.py` I get output including the line (redacted):
```
2021-06-01 17:05:15,073 - requests_oauthlib.oauth2_session - DEBUG - Supplying headers None and data {'_csrf': 'XXX, '_phase': 'authenticate', '_process': '1', 'transaction_id': 'XXX', 'cancel': '', 'identity': 'XXX', 'credential': '********', 'captcha': ''}
```
This does imply that a captcha maybe the issue. :(

GiPe66, 2021-06-01:
Yes, same problem, it seems to be related to the captcha.

sillyfrog, 2021-06-01:
I have done a proof of concept that allows this to work. But it's very much a proof of concept as we need to figure out a way to present the captcha to the user. See the PR in #21 or look at my fork at https://github.com/sillyfrog/TeslaPy

GiPe66, 2021-06-02:
It works again this morning. It seems they removed the captcha...

Les-A, 2021-06-02:
Normally use Teslajson which started failing a couple of days ago.  I installed your offering about 15 minutes ago via PIP and it works fine for me. I only have the standard authentication. Worth noting also that my program using Teslajson began working normally again at 2021-06-02 09:15:05 BST without me changing anything (it runs on a seperate PC). Maybe Tesla have (temporarily) backed out the change.

stevieb9, 2021-06-02:
It looks like they did roll out any changes. I noticed my devices working
fine this morning. I had a power outage last night so I figured a reboot
was needed, but I'm unsure of that. All I know is that it wasn't working
yesterday and it is today without me doing anything.

tdorssers, 2021-06-03:
2abb38c detects if the captcha is needed. For now it looks like it is not, so everything works as before.

closed 2021-06-08 by tdorssers