tdorssers/TeslaPy #24: Issues with credentials : Credentials rejected

88gts, 2021-07-09:
hey all, wanted to give this a shot, got everything installed but i'm having issues with logging in.  i didnt have MFA enabled, using cli.py was still requesting a captcha. i logged into tesla, configured MFA anyhow and verified the captcha works using an authenticator app.  no matter what i try, the cli.py script is returning invalid credentials. i've waited multiple iterations of credentials to verify too.  thanks!

```
Captcha: 364833
Traceback (most recent call last):
  File "cli.py", line 100, in <module>
    main()
  File "cli.py", line 61, in main
    tesla.fetch_token()
  File "/var/www/scripts/TeslaPy/teslapy/__init__.py", line 203, in fetch_token
    raise ValueError('Credentials rejected')
ValueError: Credentials rejected
```

tdorssers, 2021-07-10:
I would not expect a captcha, so this is interesting. Can you share debug output? Do you see a captcha when you login to tesla.com as well?

tdorssers, 2021-07-10:
It looks like Tesla has enabled captcha again, but not for every user agent I guess. I can authenticate successfully however.

Les-A, 2021-07-10:
I'm looking for help with understanding an error I'm getting:

> Traceback (most recent call last):
>   File "julia2.py", line 387, in <module>
>     tesla.fetch_token()
>   File "/home/pi/.local/lib/python2.7/site-packages/teslapy/__init__.py", line 180, in fetch_token
>     raise ValueError('Credentials rejected')
> ValueError: Credentials rejected

I was previously using teslajson until the problem with captcha occured a little while ago.  I then experimented with teslapy and set up a test program which was just your example code but in a while loop to log data to a file. As Tesla backed out the change I just left my original teslajson code running until captcha was reinstated a few days ago.

The thing is, that teslapy test program is still working fine.
And if I start an interactive python session and enter the teslapy statements manually, that works too - even while the other code is running (same PC and user).

I rewrote my original program to replace teslajson with teslapy and even though I cannot see anything in it that differs from my working test program (it just does a lot more with the data) it always fails.

The system I'm using is a Raspberry Pi 4 with:

> Linux rpi4 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
> Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
> [GCC 8.3.0] on linux2
> TeslaPy           1.2.0

I tried installing everything on an alternative system running::

> Linux rpi3 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
> Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
> [GCC 6.3.0 20170516] on linux2
> TeslaPy                  1.3.0

On this system I get a CAPTCHA prompt from my main program, the interactive python session and my test teslapy code.

Is it possible that the token generated for my test program when CAPTCHA was not active is still in effect?
If this is the case, will it still continue to work if I upgrade to TeslaPy 1.3.0?

Any help will be gratefully received.

Les-A, 2021-07-10:
ps: cache.json shows (I think) the bearer token was created 2021-06-19 13:23:52 and will expire 2021-08-03 13:23:52.

tdorssers, 2021-07-11:
Yes the captcha is only required when there is no bearer token in cache.json present for your email. Once the bearer token is there, the module will refresh the token automatically. So on the 3rd of August the token will be refreshed and you do not need to enter the captcha again.
Version 1.3.0 is the first version that supports captcha, and upgrading to 1.3.0 should not break anything.

dawiinci, 2021-07-11:
I am on 1.3.0 but 
```
     File "/Users/server/Library/Python/2.7/lib/python/site-packages/teslapy/__init__.py", line 170, in fetch_token
       raise ValueError('Credentials rejected')
```
What exactly do I have to do to make it work again?

tdorssers, 2021-07-11:
The most likely cause is that the captcha verification code is missing. So either you use a cache.json file with valid token, or you need to pass the authentication phase including captcha verification. Now, the captcha image will be shown in the systems default web browser. If you don't have a graphical interface, then you can use the solution presented in issue #25.

88gts, 2021-07-11:
the cli and menu scripts are no longer asking for a Captcha, but I think im blocked for some other reason now...

Max retries exceeded with url: /oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email+offline_access&state=

(Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5c92830>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',))

dawiinci, 2021-07-12:
It is not working automatically. As far as I understand I have to enter the captcha manually. But it isn't even downloaded.
This kind of breaks the functionality.

How do other Apps solve this issue? Ask the user for the captcha?

Les-A, 2021-07-12:
> Yes the captcha is only required when there is no bearer token in cache.json present for your email. Once the bearer token is there, the module will refresh the token automatically. So on the 3rd of [...]

Hi, thanks for that. Once I realised that cache.json had to be in the working directory everything fell into place and works just fine. Also, thanks for the pointer to the code for handling CAPTCHA in a non-GUI scenario.
Best regards.

88gts, 2021-07-13:
@dawiinci any luck? I am not being returned a Captcha, and not using a GUI so kinda lost here.  Is there a standalone script that creates the token?

dawiinci, 2021-07-13:
Unfortunately not. I hope there will be an integrated way at some time.

Even better would be an API without captcha. Doesn't make sense to me, because you have to refresh the token every few weeks. I think we all somehow rely on the system to just work "forever".

I am wondering how commercial apps are doing this.

tdorssers, 2021-07-13:
It appears that Tesla requires captcha only for certain User-Agents. The wrong User-Agent was sent during authentication and this is fixed in 6c0fda4. I no longer get a captcha presented with this code fix.
This module only asks for a captcha if it is a required field on the login form, which is provided by Tesla. The commercial app of Tesla doesn't get this captcha because it's a trusted User-Agent. The Tesla App only required the captcha for a few days (May 28-30 or so) and then it magically disappeared.

dawiinci, 2021-07-14:
Thanks, that sounds good.

I will test, but the change is not available with "pip install teslapy" yet.

tdorssers, 2021-07-14:
Yes this version supports the streaming API as well. I will package this version shortly and make it available on pypi.

dawiinci, 2021-08-11:
Te issue came back:

`ValueError: Credentials rejected. Captcha is required. Captcha does not match
`

omaliphant, 2021-08-12:
yep, i have this issue too... :(

tdorssers, 2021-08-12:
Please give the fix in ebcef52 a try!

dawiinci, 2021-08-14:
Unfortunately it still asks for a Captcha and opens the browser.

```
   Script Error                    tesla_connect.py: EOF when reading a line
   Script Error                    Exception Traceback (most recent call shown last):

     tesla_connect.py, line 18, at top level
     File "/Users/server/Library/Python/2.7/lib/python/site-packages/teslapy/__init__.py", line 205, in fetch_token
       form['captcha'] = self.captcha_solver(response.content)
     File "/Users/server/Library/Python/2.7/lib/python/site-packages/teslapy/__init__.py", line 319, in _solve_captcha
       return input('Captcha: ')
EOFError: EOF when reading a line
```

I just replaced the three files in teslapy folder. It should work without captchas, right?

tdorssers, 2021-08-15:
No the captcha is required at this time. The auth server 'rejects' the first login attempt and asks for a captcha, teslapy now handles this and allows successful login.

dawiinci, 2021-08-15:
I don't understand what i have to do...

Where do i have to add the captcha seen in the browser:
`import teslapy

with teslapy.Tesla(tesla_mail, tesla_pass) as tesla:
    tesla.captcha_solver = "Captcha: ????"
    tesla.fetch_token()
    vehicles = tesla.vehicle_list()
`

closed 2021-08-30 by tdorssers