tdorssers/TeslaPy #25: How to solve Captcha?

spinkb, 2021-07-10:
The docs refer to support for captcha, and sure enough I get a prompt for captcha...
But how do I solve it?
I'm using a raspberry pi with no GUI...

 can't find docs on how to handle the captcha.  Is an image saved somewhere for me to download and go view?

spinkb, 2021-07-10:
Or if I enable TOTP and use the value in my request, do I need to do that every time?  Or the refresh token will be valid for a long time not needing theMFA TOTP code for a long time?

I didn't want to enable MFA as I was afraid enabling it may cause me more problems.

tdorssers, 2021-07-10:
TOTP should work fine without a GUI. Once you are authenticated the tokens are stored on disk and refreshed automatically without user input. So you could run the module on a system with a GUI to get authenticated and then copy cache.json to your pi or you can implement a function that saves the downloaded image to a known location and pass that function to the constructor using the captcha_solver argument.
```python
def solve_captcha(svg):
    with open('captcha.svg', 'w') as f:
        f.write(svg)
    return input('Captcha: ')
```

spinkb, 2021-07-10:
Thank you, got it working!  Had to use open with 'wb' since its bytes and not text being written.

```
import teslapy

def solve_captcha(svg):
    with open('/home/TeslaPy/captcha.svg', 'wb') as f:
        f.write(svg)
        f.close()
    return input('(downlaod /home/TeslaPy/captcha.svg and reply with code:Captcha: ')

with teslapy.Tesla('email', 'pass') as tesla:
        tesla.captcha_solver = solve_captcha
        tesla.fetch_token()
        vehicles = tesla.vehicle_list()
        vehicles[0].sync_wake_up()
        charge_state = vehicles[0].get_vehicle_data()['charge_state']
        for key, value in charge_state.items():
                print(key, ' : ', value)

```

closed 2021-07-10 by spinkb