tdorssers/TeslaPy #29: Captcha issue

gaardiolor, 2021-08-09:
Hello,

Looks like tesla changed something. 

```
Traceback (most recent call last):
  File "/usr/local/scripts/teslacli/teslacli.py", line 40, in <module>
    main()
  File "/usr/local/scripts/teslacli/teslacli.py", line 31, in main
    tesla = fetch_tesla()
  File "/usr/local/scripts/teslacli/teslacli.py", line 16, in fetch_tesla
    tesla.fetch_token()
  File "/usr/local/scripts/teslacli/venv/lib/python3.9/site-packages/teslapy/__init__.py", line 210, in fetch_token
    raise ValueError('. '.join(msgs))
ValueError: Credentials rejected. Captcha is required. Captcha does not match
```

The first GET to oauth2/v3/authorize does not ask for a captcha, but after POST'ing username / password it rejects the login, and a new form is presented. Looks identical to the previous, but now a captcha is requested. teslapy doesn't handle this correctly.

Had to copy this in (__init__.py after line 198) to check for it:

```
        form = HTMLForm(response.text)
        # Retrieve captcha image if required
        if 'captcha' in form:
            response = oauth.get(self.sso_base + 'captcha')
            response.raise_for_status()  # Raise HTTPError, if one occurred
            form['captcha'] = self.captcha_solver(response.content)
            if not form['captcha']:
                raise ValueError('Missing captcha response')
            # Submit login credentials to get authorization code through redirect
            form.update({'identity': self.email, 'credential': self.password})
            response = oauth.post(self.sso_base + 'oauth2/v3/authorize',
                                  data=form, allow_redirects=False)
```

dawiinci, 2021-08-11:
#24

tdorssers, 2021-08-12:
Please give the fix in ebcef52 a try!

marcone, 2021-08-13:
Is there a way to avoid triggering the captcha altogether? When I use `cli.py` it asks for a captcha, but when I sign in with the Tesla app it does not.

omaliphant, 2021-08-13:
@tdorssers  i still seem to have an issue where the captcha doesnt save the svg file to my directory and never prompts me for the captcha response?

im using the:
`def solveCaptcha(svg):
    with open('captcha.svg', 'wb') as f:
        f.write(svg)
        f.close()
    return input('Captcha: ')`

with:
`def teslapyStartSession(creds):
    with teslapy.Tesla(creds[0], creds[2], lambda: 'XXXXXX') as tesla:
        tesla.captcha_solver = solveCaptcha
        tesla.fetch_token()
    return tesla`

is this correct?

tdorssers, 2021-08-15:
I don't know why the auth server asks for a captcha, probably because it recognizes a foreign app logging in.

tdorssers, 2021-08-15:
> @tdorssers i still seem to have an issue where the captcha doesnt save the svg file to my directory and never prompts me for the captcha response?
>
> im using the:
>
> ```
> `def solveCaptcha(svg): [...]

Does this still occur using version ebcef52?

tdorssers, 2021-08-15:
> Is there a way to avoid triggering the captcha altogether? When I use `cli.py` it asks for a captcha, but when I sign in with the Tesla app it does not.
> (edit: to be clear: this is with the latest code from github)

I haven't found a way yet, maybe it has something to do with the useragent or some other properties that are recognized by the auth server.

gaardiolor, 2021-08-17:
@tdorssers https://github.com/tdorssers/TeslaPy/commit/ebcef529f9a7754d3b025caef8721327cf2cb3a6 works, no more tracebacks. Thanks!

closed 2021-08-17 by gaardiolor

HildisviniOttar, 2021-08-21:
If you reverse proxy the native Tesla iOS app through Charles, API calls works with the Bearer tokens.  
However if you logout and try to login again, it fails and shows an NSURLConnection error. Remove proxy and it works fine. They must be hard-coding a trusted certificate in-app that Tesla API requires for initial primary app auth.

Interestingly - I wonder if the native iOS app token can be used by teslapy for "permanent" auth token. 🤔