tdorssers/TeslaPy #30: How to login with captcha

dawiinci, 2021-08-17:
Since this is not really an issue of the code - rather a missing part in the manual - I open a new one.

How do I pass the captcha to the login as seen below?

`captcha = 'btBg'
`

This part worked before:
```
with teslapy.Tesla(tesla_mail, tesla_pass) as tesla:
    tesla.fetch_token()
    vehicles = tesla.vehicle_list()
```

Thanks, I really didn't find any documentation for this.

tdorssers, 2021-08-18:
Thanks for pointing out that the docs aren't clear on this. I will look into that.
The captcha is only needed once. After successful authentication, the tokens are stored in the cache.json file. Each time the script runs, it will use the cache.json file, so you do not need to reauthenticate. You do need to supply the email at all times. So the password, passcode and captcha are not needed anymore in this case.

The captcha_solver argument must be a function. Teslapy will supply one argument to the function: the svg image. Teslapy expects a string returned by the function supplying the captcha code. The docs give this example:

```python
def solve_captcha(svg):
    with open('captcha.svg', 'wb') as f:
        f.write(svg)
    return input('Captcha: ')

with teslapy.Tesla('elon@tesla.com', 'starship') as tesla:
    tesla.captcha_solver = solve_captcha
    tesla.fetch_token()
```

Above you see an example of a function that will write the SVG to a file and gets the captcha code via stdin. To answer your question to directly supply a code, you can use a lambda (the underscore is there to ignore the argument):

```python
with teslapy.Tesla(tesla_mail, tesla_pass) as tesla:
    tesla.captcha_solver = lambda _: 'btBg'
    tesla.fetch_token()
    vehicles = tesla.vehicle_list()
```

Hope this helps.

dawiinci, 2021-08-18:
Thanks, but I still can't login.

To get the captcha in the browser I run the script without the line:
`tesla.captcha_solver = lambda _: 'btBg'
`
Btw. the captcha changes everytime I run the script.

Then I fill in the captcha and run it again but the login fails.

tdorssers, 2021-08-20:
Your example will not work. You need to use a callback function. If you want a webbrowser to display the captcha and use your indigo thing to get the code after 10 seconds, you should try this:

```python
def solve_captcha(svg):
    import webbrowser
    import tempfile
    # Use web browser to display SVG image
    with tempfile.NamedTemporaryFile(suffix='.svg', delete=False) as f:
        f.write(svg)
    webbrowser.open('file://' + f.name)
    import time
    time.sleep(10)
    return str(indigo.variables[373373398].value)

with teslapy.Tesla(tesla_mail, tesla_pass) as tesla:    
    tesla.captcha_solver = solve_captcha
    tesla.fetch_token()
    vehicles = tesla.vehicle_list()
```

`fetch_token()` will call `solve_captcha()` at the moment that the captcha code is required for login. Your code runs `fetch_token()` twice, which will be two different sessions and that will not work.

dawiinci, 2021-08-21:
It works perfectly. 

Thanks a lot!

closed 2021-08-21 by dawiinci