tdorssers/TeslaPy #31: Script asks for captcha on every login attempt

dawiinci, 2021-08-21:
As far as I understand it shouldn't do it, but even after a successful login, the next time it asks for the captcha:

```
import teslapy

def solve_captcha(svg):
    import webbrowser
    import tempfile
    # Use web browser to display SVG image
    with tempfile.NamedTemporaryFile(suffix='.svg', delete=False) as f:
        f.write(svg)
    webbrowser.open('file://' + f.name)
    indigo.server.log('#### Tesla: Login Captcha required')
    indigo.actionGroup.execute(932264821) # notification
    import time
    time.sleep(10)
    return str(indigo.variables[373373398].value)

try:
    with teslapy.Tesla(tesla_mail, tesla_pass) as tesla:
        tesla.captcha_solver = solve_captcha
        tesla.fetch_token()
        vehicles = tesla.vehicle_list()
except: 
    indigo.server.log('#### Tesla: Login failed')
```

tdorssers, 2021-08-21:
Does the script have write permission to save the cache.json file? Do you see any error logs? If you are running the script inside a docker, make sure that cache.json is stored on persistent storage. You can specify a path for the cache file.

dawiinci, 2021-08-21:
I don‘t get any errors. But this makes sense. Not sure how to specify the path. It is not inside a Docker.

tdorssers, 2021-08-22:
You can do that like this:
`with teslapy.Tesla(tesla_mail, tesla_pass, cache_file='/home/teslapy/cache.json') as tesla:`

dawiinci, 2021-08-22:
Thank you so much! It works!

closed 2021-08-22 by dawiinci