tdorssers/TeslaPy #32: ValueError: Credentials rejected. Recaptcha is required

totiherms, 2021-09-04:
Hi,

recently I'm getting this error :

Traceback (most recent call last):
  File "/home/gettoken.py", line 29, in <module>
    main(sys.argv[1],sys.argv[2])
  File "/home/gettoken.py", line 24, in main
    s = tesla.fetch_token()
  File "/opt/rh/rh-python36/root/usr/lib64/python3.6/site-packages/teslapy/__init__.py", line 227, in fetch_token
    raise ValueError('. '.join(msgs))
ValueError: Credentials rejected. Recaptcha is required

It seems that this error is raised before asking for the captcha solver? I'm using this code to solve captchas online via 2captcha, tit worked perfectlly until yesterday, and now seems that solve_captcha it's not executing

def solve_captcha(svg):
    #print("solver")
    with open('captcha.svg', 'wb') as f:
        f.write(svg)
    solver = CaptchaSolver('2captcha', api_key='xxxx')
    cairosvg.svg2png(url='captcha.svg', write_to='captcha.png')
    raw_data = open('captcha.png', 'rb').read()
    solv = solver.solve_captcha(raw_data)
    #print(solv)
    return solv

def main(user, pwd):
     with teslapy.Tesla(user, pwd) as tesla:
          tesla.captcha_solver = solve_captcha   
          s = tesla.fetch_token()
          print(tesla.token)
          return tesla.token

thanks!

MarlinMr, 2021-09-04:
Same. But I am doing major rework on my scripts, so might be that for me.

jaggerc, 2021-09-04:
Something definitely changed.  Not working for me as well.

alandtse, 2021-09-05:
To the extent it helps TeslaPy resolve this, I wanted to share my observations. It looks like recaptcha has been added today which requires JavaScript to be processed to continue. I have a python library that proxied all the requests to a user that ran into this [issue today too](https://github.com/alandtse/tesla/issues/20#issuecomment-913084360). 

If you disable javascript, which is how python API access works, it fails with the message reported above. If you enable JavaScript, the scripts checks to confirm your browser is at auth.tesla.com and fails since it's not the right location. So my conclusion is that you need to process JavaScript and also override any checks on the browser location.

tdorssers, 2021-09-05:
Captcha and recaptcha are very different. Tesla has upgraded to recaptcha, which is much harder to bypass. Any help much appreciated.

totiherms, 2021-09-05:
I'm using 2captcha to solve captchas automatically, and accepts recaptcha v2 and v3, maybe this can be a temporary solution?

https://github.com/2captcha/2captcha-python#recaptcha-v3

```
result = solver.recaptcha(sitekey='6Le-wvkSVVABCPBMRTvw0Q4Muexq1bi0DJwx_mJ-',
                            url='https://mysite.com/page/with/recaptcha',
                            version='v3',
                            param1=..., ...)
```

AGSPhoenix, 2021-09-07:
What information would be needed from 3rd party token manager apps to build a valid cache.json file? Could we get a redacted example file for those of us who have tokens already?

giachello, 2021-09-09:
Please consider signing this petition to Tesla to open up their API
https://chng.it/kfjvv9sr6k

tdorssers, 2021-09-11:
> What information would be needed from 3rd party token manager apps to build a valid cache.json file? Could we get a redacted example file for those of us who have tokens already?

This is the format:
```json
{
    "elon@tesla.com": {
        "url": "https://auth.tesla.com/",
        "sso": {
            "access_token": "***",
            "refresh_token": "***",
            "id_token": "***",
            "expires_in": 28800,
            "state": "***",
            "token_type": "Bearer",
            "expires_at": 1631332627.6740067
        },
        "ownerapi": {
            "access_token": "***",
            "token_type": "bearer",
            "expires_in": 3888000,
            "refresh_token": "***",
            "created_at": 1631303828
        }
    }
}
```

tdorssers, 2021-09-11:
I have removed headless login from the module. A web browser is now required to authenticate and manual copy-pasting of the redirected URL is needed. It's not the nicest solution, but it works (no ReCaptcha anymore) and you only need to do it once.
Please give 23038e6 a try.

AGSPhoenix, 2021-09-11:
Tried it with menu.py. Entered my email address and got taken to the sign in page. Signed in and got a 404 error.

With no advance warning that this would happen, I assumed something had broken. Figured it out when I went back to the console, but some users will not know what the 'redirect URL' is. Should probably add a brief warning about what will happen next and what to do.

That aside, seems to work great.

tdorssers, 2021-09-11:
@AGSPhoenix thanks for the feedback. f6514a6 includes a brief warning and improved docs.

tdorssers, 2021-09-13:
Packaged, released and made available on PyPI

closed 2021-09-13 by tdorssers

totiherms, 2021-09-13:
@tdorssers works perfect for me as well, thanks a lot!