tdorssers/TeslaPy #34: Just updated to latest, cron will not run scripts, need to be in directory

88gts, 2021-09-28:
hey all, seeing a very weird failure here.  i had to change my pw and re-auth, updated to 2.0 via pip3, also pulled down the repo locally.

`Successfully installed TeslaPy-2.0.0`

unfortunately, i can only get my scripts to work if in the current directory, anyone have any tips? thanks!

```
root@pigts:~# python3 /var/www/scripts/notify/TeslaPy/teslakev.py
Traceback (most recent call last):
  File "/var/www/scripts/notify/TeslaPy/teslakev.py", line 5, in <module>
    vehicles = tesla.vehicle_list()
  File "/var/www/scripts/notify/TeslaPy/teslapy/__init__.py", line 281, in vehicle_list
    return [Vehicle(v, self) for v in self.api('VEHICLE_LIST')['response']]
  File "/var/www/scripts/notify/TeslaPy/teslapy/__init__.py", line 276, in api
    return self.request('GET', uri, params=kwargs)
  File "/var/www/scripts/notify/TeslaPy/teslapy/__init__.py", line 119, in request
    response.raise_for_status()  # Raise HTTPError, if one occurred
  File "/usr/local/lib/python3.6/site-packages/requests/models.py", line 940, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 401 Client Error: Unauthorized for url: https://owner-api.teslamotors.com/api/1/vehicles
root@pigts:~# cd /var/www/scripts/notify/TeslaPy/
root@pigts:/var/www/scripts/notify/TeslaPy# python3 teslakev.py 
{
    "gps_as_of": XXX,
    "heading": 258,
    "latitude": XXX,
    "longitude": -XXX,
    "native_latitude": XXX,
    "native_location_supported": 1,
    "native_longitude": -XXX,
    "native_type": "wgs",
    "power": 0,
    "shift_state": null,
    "speed": null,
    "timestamp": 1632799082968
}
```
```
root@pigts:/var/www/scripts/notify/TeslaPy# cat teslakev.py 
import teslapy

with teslapy.Tesla('XXXXX') as tesla:
        tesla.fetch_token()
        vehicles = tesla.vehicle_list()
        vehicles[1].sync_wake_up()
        print(vehicles[1].get_vehicle_data()['drive_state'])
```

tdorssers, 2021-09-28:
When you run your script like that, the cache.json file is searched in the working directory, which is your home dir in your case. This cache.json has probably your old credentials. I can't think of any other reason for the 401. When you change the working dir to the script folder, then a different cache.json is used and the script runs without issue. I suggest you specify the directory for the cache.json file, to make sure that the correct file is used.

88gts, 2021-09-28:
Makes perfect sense, first unblocked myself by dropping cache.json in the home dir.  Then modified the declaration of cache.json in __init.py__

                 cache_file='/var/www/scripts/notify/TeslaPy/cache.json', cache_loader=None, cache_dumper=None):

Thank you for the quick response.

closed 2021-09-28 by 88gts