tdorssers/TeslaPy #35: cache.json portability?

morpheus65535, 2021-09-30:
Hey, first, I would like to thank you for this great piece of software! I really enjoyed it until Tesla broke our ability to simply connect to their API using a Python script. Scheduling my climate control and heated seats to be hot when I get into my car in the winter morning of northern Canada was a great addition.

More of a question here but, now that we have to live with the reCaptcha and a real browser to get a valid token, I was asking myself if there's some way of reusing the cache.json file generated on my laptop inside a docker container. Basically, from what I've tested, I can't reuse it.

Am I doing something wrong or is there something store inside the token that prevent it from being reused on another computer?

Thanks for your help!

MarlinMr, 2021-09-30:
Putting the cache.json in the working directory and referencing it in the script works for me. 

```
#!/usr/bin/python3                                                                                                                                                                                                                                                                        
import teslapy, math, json                                                                                                                                                                                                                                                                
                                                                                                                                                                                                                                                                                          
tesla_config_file = '/home/user/.config/tesla/tesla.json'                                                                                                                                                                                                                                 
tesla_cache_file = '/home/user/.config/tesla/cache.json'
mile = 1.609344

with open(tesla_config_file) as f:
        data = json.load(f)

with teslapy.Tesla(email=data["email"],password=data["password"], cache_file=tesla_cache_file) as tesla:
        tesla.fetch_token()
        vehicles = tesla.vehicle_list()
        vehicle = vehicles[0]
```

Just put the option `cache_file=<file_name>` in your `teslapy.Tesla()`

morpheus65535, 2021-09-30:
@MarlinMr thanks for referencing that argument. I was under the impression that it was created/loaded alongside the python script implementing the Tesla class but it seems to be best to specify a location by myself.

closed 2021-09-30 by morpheus65535