tdorssers/TeslaPy #37: Get Safety Score endpoints data

rog0978, 2021-10-13:
Anyone managed to get Safety Score data? I see the end points in endpoints.json, but they don't seem to work with e.g. the cli.py example.

philipsd6, 2021-11-05:
FYI, you need to add `timezone` to the params, otherwise you get a validation error.

```python
...
    print(tesla.get(url, params={'vin': vehicles[0]['vin'],
                                 'deviceLanguage': 'en',
                                 'deviceCountry': 'US',
                                 'timezone': 'UTC'}))
```

tdorssers, 2021-11-07:
Thanks, I have added this example to the readme!

philipsd6, 2021-11-07:
yw! Now the only thing I'm struggling to figure out is what the numbers mean in the `dailyAggregation.metrics`. Seems like the Deciles are a float between 0 and 1, and they are opposite of what I understand decile rankings to be, and I guess it appears to be a ranking against all other drivers? The Median... the median of what? Same for Average -- average of what? Are these numbers aggregated by an unknown number of measurements taken during that daily period? I dunno if I did it right, but I made some graphs out of the data -- but if anyone has a better sense of the data, maybe that can be shared!? :)

andeplane, 2021-12-10:
I'm getting a 400 on this request, but maybe it's because I'm not in the US and my car isn't measuring safety score?

tdorssers, 2021-12-10:
Yes, as far as I know it's US only.

closed 2022-03-04 by tdorssers