tdorssers/TeslaPy #44: No login possible

dawiinci, 2021-11-03:
```
import teslapy

def solve_captcha(svg):
    import webbrowser
    import tempfile
    # Use web browser to display SVG image
    with tempfile.NamedTemporaryFile(suffix='.svg', delete=False) as f:
        f.write(svg)
    webbrowser.open('file://' + f.name)
    indigo.server.log('#### Tesla: Login Captcha required')
    indigo.actionGroup.execute(932264821) # notification
    import time
    time.sleep(10)
    return str(indigo.variables[64356346].value)

try:
    with teslapy.Tesla(tesla_mail, tesla_pass, cache_file='/Users/server/Skripte/TeslaPy/cache.json') as tesla:
        tesla.captcha_solver = solve_captcha
        tesla.fetch_token()
        vehicles = tesla.vehicle_list()
except: 
    indigo.server.log('#### Tesla: Login failed')
```

This code worked, but since yesterday it fails to login. No browser is opened for the captcha though. Any ideas?

tdorssers, 2021-11-03:
This login method is no longer supported. Tesla has replaced captcha by recaptcha. Some module maintainers have decided to use a paid recaptcha solver service to support headless login. I decided to drop support for headless login and implement authentication though a real web browser or WebView. Please take a look at the readme for more information.

dawiinci, 2021-11-04:
Thanks, but I barely brought this to work. I just don't get it working with the new method...

There is no captcha in the readme.

tdorssers, 2021-11-05:
There is no `captcha_solver` parameter anymore for the `Tesla` class. The recaptcha will be shown by the browser controlled by selenium.

closed 2021-11-28 by tdorssers