tdorssers/TeslaPy #45: Asynchronous authenticator?

morpheus65535, 2021-11-14:
It would be nice to be able in a web application to show the auth url then wait for the user to provide the url including the code through a form submission. Any idea how I could achieve that?

tdorssers, 2021-11-16:
By using an `authenticator` function, which is a callback, you get some async functionality. TeslaPy is a synchronous library. Which framework are you planning to use?

morpheus65535, 2021-11-16:
Currently I'm running some test with Flask and Threading.Event. Looks promising and I'll update this issue when I get it working for the benefit of others.

tdorssers, 2021-11-17:
I am working on splitting `fetch_token()` in two methods. If you don't supply an argument to fetch_token, then it will behave the same as before. A new method will generate the auth url and you can supply the redirected url as an argument to fetch_token.
Will that help/make life easier?

morpheus65535, 2021-11-17:
Will those two method be independant? Is it gonna be possible to call the first method to get the auth_url then, from a different Falsk route, call the fetch_token method with only the auth_url and (let's call it) the code_url as argument? How will the Tesla class will behave on init if there's no token?

tdorssers, 2021-11-17:
Yes the methods are independent and will behave like you described. On init, the class will try to read the token from the cache file. If there is no token in the cache file or no file at all, then nothing will happen. The first API call that you try to make, will call fetch_token() without arguments automatically, meaning it will use the authenticator function to get the code. So in your Flask app, you need to make sure not to call an API before you are authenticated using the 'two step' way.

tdorssers, 2021-11-17:
Please check out 333c114. You can use it like this:

```python
import teslapy

with teslapy.Tesla('elon@tesla.com') as tesla:
    print(tesla.authorization_url())
    tesla.fetch_token(authorization_response=input('URL: '))
    vehicles = tesla.vehicle_list()
    print(vehicles[0])
```

tesla.authorization_url() will generate the auth url.
you pass the redirected url to fetch_token() using keyword argument authorization_response.

morpheus65535, 2021-11-17:
Does it need to be call inside a context manager or can I simply instantiate the tesla object and reuse it?

tdorssers, 2021-11-18:
Context manager is optional. You can instantiate a tesla object and close it when done. authorization_url() and fetch_token() must be called on the same instance.
```python
tesla = Tesla('elon@tesla.com')
# do stuff
tesla.close()
```

morpheus65535, 2021-11-18:
It works perfectly fine the way you did it! Thank you very much! 👍

morpheus65535, 2021-11-18:
Would it be too much if I ask for a `logout` method that would set `authorized` to `False` and delete the cached token?

tdorssers, 2021-11-18:
No not really, please give 917f6af a try.

tdorssers, 2021-11-20:
Thanks! How are you managing with vehicle wake up and other calls from your web app?

morpheus65535, 2021-11-20:
Basically my web app is an evolution from my headless shceduled preheating app that was configured using environment variables. I'm refactoring it to use a config file instead and the frontend will be used to generate the token and edit the config file.

So, until the token is valid and teslapy autorized is True, the schedule task isn't able to start. Once the user as authenticated and we have a valid token, the scheduled task will run based on user requirement (ex.: every weekday at 7am). I also have a maximum preheating length that stop the preheating after a specified delay to prevent preheating to be left enabled for 4 hours if not required.

morpheus65535, 2021-11-26:
@tdorssers any ETA for releasing master to a new version on pypi? For now I'm linking to this repo in my requirements.txt but making it more "official" would be great ;-)

thanks!

tdorssers, 2021-11-28:
Just released 2.2.0 a few minutes ago!

morpheus65535, 2021-11-28:
Thank you very much! Really appreciated. :-)

I'll close this issue now.

closed 2021-11-28 by morpheus65535