tdorssers/TeslaPy #51: HTTPError: 401 Client Error: Unauthorized for url: https://owner-api.teslamotors.com/api/1/vehicles

Les-A, 2022-01-12:
I've had a data collection program running for many months now without issue and has picked up successfully with the cached tokens after every restart. Today it failed with the following:

> Traceback (most recent call last):
>   File "/mnt/seagate/private/Python/tesla/Julia.py", line 393, in <module>
>     vehicles = tesla.vehicle_list()
>   File "/usr/local/lib/python2.7/dist-packages/teslapy/__init__.py", line 384, in vehicle_list
>     return [Vehicle(v, self) for v in self.api('VEHICLE_LIST')['response']]
>   File "/usr/local/lib/python2.7/dist-packages/teslapy/__init__.py", line 379, in api
>     return self.request('GET', uri, params=kwargs)
>   File "/usr/local/lib/python2.7/dist-packages/teslapy/__init__.py", line 148, in request
>     response.raise_for_status()  # Raise HTTPError, if one occurred
>   File "/home/pi/.local/lib/python2.7/site-packages/requests/models.py", line 940, in raise_for_status
>     raise HTTPError(http_error_msg, response=self)
> requests.exceptions.HTTPError: 401 Client Error: Unauthorized for url: https://owner-api.teslamotors.com/api/1/vehicles

I also tried using another program which is closely based on one of your sample programs and it fails with same error.

I updated to the very latest TeslaPy (2.3.0) but no change.

The token was refreshed on 2022-01-07 19:54:58 and has an expiry of 2022-02-21 19:54:58 (dates are in UK format).

As this has just started to fail I'm not sure if this is a temporary fault with the Tesla infrastructure or if there has been a significant change again.

The official Tesla App is still working correctly at the moment.

Would appreciate any advice on how to further diagnose this problem.

hawkeye217, 2022-01-13:
Same thing for me today but with the ``products`` endpoint.

```
requests.exceptions.HTTPError: 401 Client Error: Unauthorized for url: https://owner-api.teslamotors.com/api/1/products
```

tdorssers, 2022-01-13:
If you were running a version before 2.1, then the JTW (or some people call it refresh token), which expires every 45 days or so, was used to access API endpoints. Since 2.1 the SSO token which expires every 8 hours is used, like the official App does. In your first message you showed a traceback of an older version of TeslaPy, if I'm not mistaken. I just ran a test and token refresh is still working for me. Please try to login again using the browser method and see if it starts working from there.
In your second message you posted a traceback of the oauthlib itself, which complains that an expired token has been added. I have never seen such a message. Which package versions are you using of requests_oauthlib and oauthlib?
Please turn on debug logging and share the output of the debug logs, but masquerade your tokens shown in the logs before sharing.

JuhaSeppa, 2022-01-13:
I'm seeing the same problem overnight as my automation tries to charge Tesla Powerwall:
`401 Client Error: Unauthorized for url: https://owner-api.teslamotors.com/api/1/products`

If I call fetch_toke() directly I get
`403 Client Error: Forbidden for url: https://auth.tesla.com/oauth2/v3/authorize`

tdorssers, 2022-01-13:
Have you also never used the browser method to authenticate and still using the cached token, which was acquired using the previous headless login methods (TeslaPy 1.x)?

JuhaSeppa, 2022-01-13:
I have used the cached token only so far.

B-Hartley, 2022-01-13:
@JuhaSeppa  - I'm having similar problems using the TeslaGateway integration in home assistant to change powerwall status.
Assume it's something to do with the Tesla security incident over the last couple of days.

Hoping for a fix.

HeraldCoupe, 2022-01-13:
> @JuhaSeppa - I'm having similar problems using the TeslaGateway integration in home assistant to change powerwall status. Assume it's something to do with the Tesla security incident over the last couple of days.
> 
> Hoping for a fix.

Likewise, hoping for a fix soon.

Les-A, 2022-01-13:
> If you were running a version before 2.1, then the JTW (or some people call it refresh token), which expires every 45 days or so, was used to access API endpoints. Since 2.1 the SSO token which [...]

Thanks for the quick response. Yes, I didn't upgrade the version last time and am relying on the longer expiry time of the cached token. I'll try to re-login today and get back to you with the results. Having read through the instructions I'm reasonably confident I can do that.

88gts, 2022-01-13:
I was faced with the same issue so I pulled down the latest and can't re-auth using menu.cli.  I get the URL and login but the loading bar runs through the 30 seconds of time and never reaches the Page Not Found at success.

tdorssers, 2022-01-13:
Please try another web browser or use one of the two suggested automated methods: pywebview or selenium

88gts, 2022-01-13:
Was able to re-auth using another system.   Thanks!

Les-A, 2022-01-13:
> Hi, I've updated everything now, brought it into line with your examples and managed to generate a new token. 
 As of now the versions of requests_oauthlib and oauthlib according to pip install appear to be:

python-oauthlib  3.0.1
python-requests-oauthlib 1.2.0
python3-oauthlib 3.1.0
python3-requests-oauthlib 1.3.0

For some reason under python 2 I'm still picking up the old version of teslapy and I'm not sure why, but it is working normally under python 3 so I'm going to stick with that. So as far as I am concerned the problem is user error and this can be closed.

Many thanks for your help.

JuhaSeppa, 2022-01-14:
Latest version of TeslaPy and browser method of authentication works. Thanks!

tdorssers, 2022-01-22:
Added the option to use a refresh token obtained by 3rd party authentication apps as an alternative to the built in browser method. Check out the latest commit.

closed 2022-03-04 by tdorssers