tdorssers/TeslaPy #52: menu.py: TypeError: unsupported format string passed to NoneType.__format__

ldrolez, 2022-01-13:
With the git version, menu.py starts displaying data, and then encounters an error:

```
...
...
Driver Seat Heater: 0                  Passenger Seat Heater: 0
Is Front Defroster On: False           Is Rear Defroster On: False
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "./menu.py", line 329, in <module>
    main()
  File "./menu.py", line 312, in main
    menu(vehicles[idx])
  File "./menu.py", line 176, in menu
    show_vehicle_data(vehicle.get_vehicle_data())
  File "./menu.py", line 61, in show_vehicle_data
    print(fmt.format(ve['vehicle_name'], vehicle.dist_units(ve['odometer'])))
TypeError: unsupported format string passed to NoneType.__format__
```

 $ python3 --version
Python 3.7.3

tdorssers, 2022-01-13:
That would mean that either ve['vehicle_name'] or ve['odometer'] equals to None, which is unexpected.
Can you add this line before line 61 and share the output:
```print(ve['vehicle_name'], ve['odometer'])```

ldrolez, 2022-01-13:
Yes I never tried to change the name of the car

tdorssers, 2022-01-13:
Please checkout menu.py of 8e0ec2e

ldrolez, 2022-01-17:
Great! it works

closed 2022-01-17 by ldrolez