tdorssers/TeslaPy #54: crash when vehicle is in service

npitre, 2022-01-22:
The code fails when vehicle is in service and awake.

For example, selecting 3 at the menu.py prompt to wake up the vehicle produces
this:

Choice (0 to quit): 3
--------------------------------------------------------------------------------
Please wait...
2022-01-21 21:42:54,913 - teslapy - INFO - Corbimobile is asleep
2022-01-21 21:43:12,121 - teslapy - INFO - Corbimobile is online
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "TeslaPy/./menu.py", line 332, in <module>
    main()
  File "TeslaPy/./menu.py", line 315, in main
    menu(vehicles[idx])
  File "TeslaPy/./menu.py", line 173, in menu
    if not vehicle.mobile_enabled():
  File "TeslaPy/teslapy/__init__.py", line 479, in mobile_enabled
    return self.tesla.get(uri)['response']
  File "/usr/lib/python3.9/site-packages/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "TeslaPy/teslapy/__init__.py", line 121, in request
    response.raise_for_status()  # Raise HTTPError, if one occurred
  File "/usr/lib/python3.9/site-packages/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 405 Client Error: vehicle is currently in service for url: https://owner-api.teslamotors.com/api/1/vehicles/1492************/mobile_enabled

Restarting menu.py crashes right away until the vehicle goes back to sleep.

tdorssers, 2022-01-22:
Please checkout f268dda. This patch should fix this crash.

npitre, 2022-01-22:
Yes, it is fixed ...

... ntil option 4 is used:

```
Choice (0 to quit): 4
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "TeslaPy/./menu.py", line 335, in <module>
    main()
  File "TeslaPy/./menu.py", line 318, in main
    menu(vehicles[idx])
  File "TeslaPy/./menu.py", line 209, in menu
    show_charging_sites(vehicle)
  File "TeslaPy/./menu.py", line 128, in show_charging_sites
    sites = vehicle.get_nearby_charging_sites()
  File "TeslaPy/teslapy/__init__.py", line 500, in get_nearby_charging_sites
    return self.api('NEARBY_CHARGING_SITES')['response']
  File "TeslaPy/teslapy/__init__.py", line 441, in api
    return self.tesla.api(name, {'vehicle_id': self['id_s']}, **kwargs)
  File "TeslaPy/teslapy/__init__.py", line 322, in api
    return self.request(endpoint['TYPE'], uri, serialize,
  File "TeslaPy/teslapy/__init__.py", line 131, in request
    response.raise_for_status()  # Raise HTTPError, if one occurred
  File "/usr/lib/python3.9/site-packages/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 405 Client Error: vehicle is currently in service for url: https://owner-api.teslamotors.com/api/1/vehicles/1492************/nearby_charging_sites
```

tdorssers, 2022-01-23:
c8949dc should fix both cases. Can you verify this?

npitre, 2022-01-23:
Yes, this is fixed, thanks.

FYI:

One thing is weird. The reported temperature is definitively current (it 
is coherent with the car being parked outside).

However the battery state of charge doesn't reflect what the Tesla 
app shows. In the app it is 71% and slowly dropping 1% each day or so, 
whereas TeslaPy reports 78% which was the state of charge the day the 
car was dropped off at the service center.

npitre, 2022-01-24:
OK, here's another for you. The car moved (hopefully being serviced 
at last) and now I get:

```
Select vehicle: 0
[...]
Vehicle is in service
[...]
Driver/Pass Front Door: Closed/Closed  Driver/Pass Rear Door: Closed/Closed
Drvr/Pass Front Window: Open/Closed    Driver/Pass Rear Window: Closed/Closed
Traceback (most recent call last):
  File "TeslaPy/./menu.py", line 337, in <module>
    main()
  File "TeslaPy/./menu.py", line 320, in main
    menu(vehicles[idx])
  File "TeslaPy/./menu.py", line 181, in menu
    show_vehicle_data(vehicle.get_vehicle_data())
  File "TeslaPy/./menu.py", line 75, in show_vehicle_data
    print(fmt.format(door[ve['ft']], door[ve['rt']]))
IndexError: list index out of range
```

So I printed `ve` right before that and got:

```
    "df": 0,
    "dr": 0,
    "fd_window": 2,
    "fp_window": 0,
    "ft": 16,
```

That `"ft": 16,` looks weird. That must be really really opened when
in service.  ;-)

tdorssers, 2022-01-25:
Haha yes something link that. Anyway 59cde03 should fix the traceback.

npitre, 2022-01-25:
OK thanks.

Anyway the car won't wake up now (as expected) and I doubt I'll be able 
to catch another weird value. At least we know that the ft attribute 
might be 16 sometimes.

closed 2022-01-25 by npitre