tdorssers/TeslaPy #58: No module named 'teslapy'

elic1973, 2022-02-07:
Hi,

Running py file with SSH works perfectly.
Running the same py file from HA using shell_command results with : No module named 'teslapy' - fails on the import command

My config:
shell_command:
  enable_sentry: cd /config/scripts && python3 enable_sentry.py

enable_sentry.py file
try:
    import teslapy
except Exception as Argument:
     with open('logerror.txt', 'w') as f:
        f.write(str(Argument))

System Health
version: core-2022.2.3
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
user: root
virtualenv: false
python_version: 3.9.7
os_name: Linux
os_version: 5.10.63-v8
arch: aarch64

host_os: Home Assistant OS 7.2
update_channel: stable
supervisor_version: supervisor-2022.01.1
docker_version: 20.10.9
disk_total: 109.3 GB
disk_used: 34.1 GB
healthy: true
supported: true
board: rpi4-64
supervisor_api: ok
version_api: ok
installed_addons: Samba share (9.5.1), Mosquitto broker (6.0.1), Terminal & SSH (9.3.0), File editor (5.3.3), Duck DNS (1.14.0), Frigate NVR (2.3), PostgreSQL (1.0.0), Grafana (7.4.1), TeslaMate (0.17.1)

tdorssers, 2022-02-07:
I believe that you need to install TeslaPy in the homeassistant container context first.

elic1973, 2022-02-07:
> I believe that you need to install TeslaPy in the homeassistant container context first.

I managed to get additional information. 
Running from HA I get for the sys.path : /usr/**local/**lib/python3.9/site-packages
Running the same script from SSH I get for the sys.path : /usr/lib/python3.9/site-packages

I also tried to install with -t to /usr/local/lib/python3.9/site-packages with no success.

Now how could I overcome this? not much of a guru if you can explain how to install it in HA container...

closed 2022-02-09 by elic1973