tdorssers/TeslaPy #59: My looping app will run for hours but never a day

israndy, 2022-02-26:
My code is published as: https://github.com/israndy/SuperchargerLog

I continue to receive this error, and because it happens daily I am not sure how to track it down, I just see this output after the last successful step the program did run:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()

And it goes on for page after page of errors.

israndy, 2022-02-27:
Same thing with my original app TesSense: https://github.com/israndy/TesSense

This app doesn't even need to wake the Tesla to fail, ran from 10:30 last night and by just after 2 am it had crashed, rather typical

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 268, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1241, in recv_into
    return self.read(nbytes, buffer)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1099, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/adapters.py", line 440, in send
    resp = conn.urlopen(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 785, in urlopen
    retries = retries.increment(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/packages/six.py", line 770, in reraise
    raise value
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 451, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 340, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='owner-api.teslamotors.com', port=443): Read timed out. (read timeout=10)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/israndy/Documents/Xcode/Python3/Tessense.py", line 62, in <module>
    try : car = vehicles[0].get_vehicle_summary()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/teslapy/__init__.py", line 408, in get_vehicle_summary
    self.update(self.api('VEHICLE_SUMMARY')['response'])
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/teslapy/__init__.py", line 404, in api
    return self.tesla.api(name, {'vehicle_id': self['id_s']}, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/teslapy/__init__.py", line 286, in api
    return self.request(endpoint['TYPE'], uri, serialize,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/teslapy/__init__.py", line 113, in request
    response = super(Tesla, self).request(method, url, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests_oauthlib/oauth2_session.py", line 515, in request
    return super(OAuth2Session, self).request(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/sessions.py", line 529, in request
    resp = self.send(prep, **send_kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/sessions.py", line 645, in send
    r = adapter.send(request, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/adapters.py", line 532, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='owner-api.teslamotors.com', port=443): Read timed out. (read timeout=10)

israndy, 2022-03-03:
OK, I indented my code in one app (Tessense.py) starting with the While as you suggested and ran for the last few days w/o issue. But I continue to have issues with SuperchargerLog.py, so I'll continue to look at the code to see what the differences are, it's probably just a typo

israndy, 2022-03-04:
> Closed #59 <https://github.com/tdorssers/TeslaPy/issues/59> via 1ee8998 <https://github.com/tdorssers/TeslaPy/commit/1ee899875032e2d4b9eabcfbdba60ea313dd3d7c>.

Still having the issue in the app (https://github.com/israndy/SuperchargerLog <https://github.com/israndy/SuperchargerLog>) I asked about. It rarely runs for more than a few hours w/o a TeslaPy issue.  My TesSense.py app seems very stable so I am adding code from this app until that one starts to break. When I saw your email this morning I ran SuperchargerLog.py and it didn’t even make an hour.:

***@***.*** Python3 % python3 SuperchargerLog.py

Waking your car and logging local Superchargers every 5 minutes

Starting connection to Tesla...
Rosebud is online
Local Superchargers:
 ['2022-03-04 06:57', 'Alameda', '10', '12', 'Oakland', '12', '16', 'Oakland', '12', '12', 'Emeryville', '9', '11']
Local Superchargers:
 ['2022-03-04 07:02', 'Alameda', '10', '12', 'Oakland', '12', '16', 'Oakland', '12', '12', 'Emeryville', '9', '11']
Local Superchargers:
 ['2022-03-04 07:07', 'Alameda', '11', '12', 'Oakland', '11', '16', 'Oakland', '12', '12', 'Emeryville', '9', '11']

68%

Local Superchargers:
 ['2022-03-04 07:12', 'Alameda', '11', '12', 'Oakland', '14', '16', 'Oakland', '12', '12', 'Emeryville', '9', '11']
Local Superchargers:
 ['2022-03-04 07:17', 'Alameda', '11', '12', 'Oakland', '14', '16', 'Oakland', '12', '12', 'Emeryville', '10', '11']
Local Superchargers:
 ['2022-03-04 07:22', 'Alameda', '12', '12', 'Oakland', '13', '16', 'Oakland', '12', '12', 'Emeryville', '10', '11']
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 268, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1241, in recv_into
    return self.read(nbytes, buffer)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1099, in read
    return self._sslobj.read(len, buffer)
ConnectionResetError: [Errno 54] Connection reset by peer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/adapters.py", line 440, in send
    resp = conn.urlopen(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 785, in urlopen
    retries = retries.increment(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/packages/six.py", line 769, in reraise
    raise value.with_traceback(tb)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 268, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/socket.py", line 704, in readinto
[... 5 lines cut ...]
urllib3.exceptions.ProtocolError: ('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/israndy/Documents/Xcode/Python3/SuperchargerLog.py", line 35, in <module>
    car = vehicles[0].get_vehicle_summary()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/teslapy/__init__.py", line 444, in get_vehicle_summary
    self.update(self.api('VEHICLE_SUMMARY')['response'])
[... cut ...]
    resp = self.send(prep, **send_kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/sessions.py", line 645, in send
    r = adapter.send(request, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/adapters.py", line 501, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer'))
***@***.*** Python3 %

tdorssers, 2022-03-04:
Ok, I'm running your app now. Let's see if I can reproduce the issue.

israndy, 2022-03-04:
Thanks so much, I am on a Mac running the latest OS and having installed XCode to get my updated Python 3.9.1

I also get this error occasionally:

Traceback (most recent call last):
  File "/Users/israndy/Documents/Xcode/Python3/SuperchargerLog.py", line 76, in <module>
    output.append( str(site['available_stalls']) )
KeyError: 'available_stalls'

Not sure what would cause this error as it didn’t choke on the site['name'] which is part of the same structure.

tdorssers, 2022-03-05:
To get rid of KeyError change the line to: output.append( str(site.get('available_stalls', 0)) )
Some objects disappear instead of returning zero, depending how Tesla implemented the API call.

'Connection reset by peer' usually indicates some kind of network connectivity issue. Is your Mac using a wireless connection?

I have run your app for 10 hours without issue:

Waking your car and logging local Superchargers every 5 minutes

Starting connection to Tesla...
Tim's Tesla is asleep
Waking...

Opened New Superchargers Log File on Superchargers.csv 

Local Superchargers:
 ['2022-03-04 20:45', 'Breda', '7', '16', 'Oosterhout', '8', '12', 'Dordrecht', '8', '16', 'Tilburg', '14', '16']
Local Superchargers:
 ['2022-03-04 20:50', 'Breda', '14', '16', 'Oosterhout', '6', '12', 'Dordrecht', '13', '16', 'Tilburg', '16', '16']
Local Superchargers:
 ['2022-03-04 20:55', 'Breda', '14', '16', 'Oosterhout', '6', '12', 'Dordrecht', '12', '16', 'Tilburg', '16', '16']
Waking...

58%

Local Superchargers:
 ['2022-03-04 21:01', 'Breda', '14', '16', 'Oosterhout', '6', '12', 'Dordrecht', '12', '16', 'Tilburg', '16', '16']
Local Superchargers:
 ['2022-03-04 21:06', 'Breda', '14', '16', 'Oosterhout', '6', '12', 'Dordrecht', '13', '16', 'Tilburg', '16', '16']
Local Superchargers:
 ['2022-03-04 21:11', 'Breda', '15', '16', 'Oosterhout', '7', '12', 'Dordrecht', '14', '16', 'Tilburg', '16', '16']

<snip>

Waking...
Local Superchargers:
 ['2022-03-05 06:43', 'Breda', '16', '16', 'Oosterhout', '10', '12', 'Dordrecht', '16', '16', 'Tilburg', '15', '16']
Local Superchargers:
 ['2022-03-05 06:48', 'Breda', '16', '16', 'Oosterhout', '10', '12', 'Dordrecht', '16', '16', 'Tilburg', '15', '16']
Traceback (most recent call last):
  File "C:\Users\tim\TeslaPy\supercharger.py", line 103, in <module>
    sleep( 300 )
KeyboardInterrupt

tdorssers, 2022-03-05:
You can also specify retry and timeout params to make it more robust:

with teslapy.Tesla( username, retry=3, timeout=20 ) as tesla:

israndy, 2022-03-05:
Thanks, added that to the SuperchargerLog.py code and restarted it just now.

I had simply moved the constant syncwakeup call into my stable app and got the same crash yesterday if that’s helpful.

israndy, 2022-03-06:
Yip, first time the script ran overnight, very nice, I thing the retry and time-out trick did it, thanks!

tdorssers, 2022-03-06:
Ok, indenting and enabling retries and increased timeout made your script handle connection issues, so it doesn't crash anymore. Nice.

tdorssers, 2022-03-10:
Ok the 500+ errors are not about connectivity between your script and Tesla's servers, but between Tesla's servers and your car. Teslapy will not retry by default for those errors, but you can influence that:

retry = teslapy.Retry(total=3, status_forcelist=(500, 502, 503, 504))
with teslapy.Tesla(username, retry=retry, timeout=20) as tesla:

TeslaPy does not support asyncio, but you can use threading instead.

israndy, 2022-03-10:
Thanks for the reply. Is the forcelist just errors that it will retry when it sees them? My biggest concern is to keep the app from quitting when it hits a non-lethal error. 

I tried your previous retry/timeout change but the app started locking up when it sent sync_wakeup. I waited hours for it to continue. I ended up pulling out that code, back to where I started.

Sadly, I have apparently locked myself out of my car, perhaps too much accessing it from an app. I can’t even get in with my phone with the WiFi turned off, it does still work as a key of course. I’ll try again after a day or so.

The asyncio is not for TeslaPy, but to inform my Sense Energy Monitor of the power the Tesla is using to charge, since there is no way for it to detect it directly I am hoping to figure out how to send it a datapacket with the info as though my app were another IoT device.

tdorssers, 2022-03-10:
Yes as described here: https://urllib3.readthedocs.io/en/latest/reference/urllib3.util.html
The downside of retrying on those HTTP codes is that if you do that too often, then the Tesla WAF will block your account for a couple of hours, because it's like a denial of service attack.

israndy, 2022-03-10:
Is there another way to detect those codes and just go to sleep for a time instead of having the app exit or forcing a retry that may boot me offline?

tdorssers, 2022-03-11:
Yes all those codes are raised as a TeslaPy.HTTPError exception and you can catch that exception and do your magic.

closed 2022-03-19 by tdorssers

israndy, 2022-07-25:
I continued to have issues with the app over the months since last posting, but since the app TesSense is tracking free solar and controlling the charing of the car based on that it didn't make sense for the app to be running all night which is usually when I had issues, perhaps they were Tesla doing maintenance or too many cars plugging in so I added a line to sleep the app from 8pm to 8am and it's gotten super stable.

commit 1ee89987 (2022-03-04): Disable API requests after close #59