tdorssers/TeslaPy #60: Gett

aviadoffer, 2022-03-01:
I disabled MFA on my tesla account yet I still can't run menu.py

Here is the error I get:

Enter email: MY_TESLA_EMAIL
Traceback (most recent call last):
  File "menu.py", line 336, in <module>
    main()
  File "menu.py", line 306, in main
    vehicles = tesla.vehicle_list()
  File "C:\Users\aviad\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\teslapy\__init__.py", line 327, in vehicle_list
    return [Vehicle(v, self) for v in self.api('VEHICLE_LIST')['response']]
  File "C:\Users\aviad\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\teslapy\__init__.py", line 312, in api
    self.fetch_token()
  File "C:\Users\aviad\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\teslapy\__init__.py", line 183, in fetch_token
    url = self.authorization_url()
  File "C:\Users\aviad\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\teslapy\__init__.py", line 162, in authorization_url
    response.raise_for_status()  # Raise HTTPError, if one occurred
  File "C:\Users\aviad\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\requests\models.py", line 960, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 500 Server Error: Internal Server Error for url: https://auth.tesla.com/oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email+offline_access&state=X2yy3x9XLOKcGwS3WQzW7aSRRUh9CL&code_challenge=sHsqezDB3WXgqipH5Cx8jDcx1YTwEQtV3hjHVEj7k3I&code_challenge_method=S256&login_hint=MY_TESLA_EMAIL

aviadoffer, 2022-03-01:
I think in my case it's not oauth2/v3 but oauth2/v1 ...

8VRST, 2022-03-02:
> I think in my case it's not oauth2/v3 but oauth2/v1 ... I also have Tesla Solar so maybe it's related.

Did you solve the error? Got the same right now.

aviadoffer, 2022-03-02:
No if I switch to v1 it doesn't work. I don't know whats going on here

tdorssers, 2022-03-02:
It looks like Tesla has changed something. I get the same issue when I reauthenticate. A refresh token still works though.

aviadoffer, 2022-03-02:
@tdorssers  does refresh token prompts via the webbrowser or is it done by the tesalpy ? meaning is it interactive or not ?

tdorssers, 2022-03-02:
I mean if you already have a refresh token then you can use it like this: https://github.com/tdorssers/TeslaPy#3rd-party-authentication-apps

It looks like Tesla has restricted the version (v1 or v3), client_id (ownerapi or ownership) and redirect_uri combinations. The combination that used to work, doesn't anymore.

mikkel75, 2022-03-18:
I have tried the latest code and still get 

"You don't have permission to access "http://auth.tesla.com/oauth2/v3/authorize?" on this server."

When I try to login in my normal browser, after email is entered it redirects to:

https://auth.tesla.com/oauth2/v1/authorize...

for password validation.

Any chance you can support this auth site? Seems like you have hard-coded "/oauth2/v3/" in the code...

Thanks,
M

tdorssers, 2022-03-19:
@mikkel75 did you use the latest version on pipy or did you clone this repo? As far as I know, the owner API requires a v3 token.

mikkel75, 2022-03-19:
@tdorssers thanks for reply.
I cloned this repo, do I need to do something else?

tdorssers, 2022-03-19:
Can you turn on debug logging and share the output?
The version that is on pypi should work again according to https://github.com/timdorr/tesla-api/issues/538, so I might revert the changed I made...

mikkel75, 2022-03-20:
@tdorssers Sorry, now it seems to work actually...don't know what has changed...

tdorssers, 2022-03-21:
@mikkel75 No worries

carrotcn, 2022-03-29:
Same issue here.. The version on pipy and via repo clone give the same error in the Chrome browser pop up after entering password:

"Access Denied
You don't have permission to access "http://auth.tesla.com/oauth2/v3/authorize?" on this server."

First time user so I don't have a refresh token..

tdorssers, 2022-03-30:
I can't reproduce this. Can you manually try to remove the login_hint parameter from the url? Or check out version [33db201](https://github.com/tdorssers/TeslaPy/commit/33db201d7c9e2c78616d3acf5063c19e11aa8130) which doesn't put login_hint in the url.

carrotcn, 2022-03-30:
I tried version [33db201] but the error has been the same. The only difference it made with the removal of login_hint was that in the pop up browser window it asked me to key in the email address again. In the latest version with login_hint the email address was passed from the terminal input to the browser window.

It should not be related to the Tesla account being used - I created a new Tesla account (without vehicles in it of course) but the error has been the same.

tdorssers, 2022-03-30:
Are you able to login to the official Tesla App using that account?
Do you get the 403 when you entered your password, or directly when opening the URL?
Have you tried 3rd party token generation apps https://teslascope.com/help/generating-tokens?

carrotcn, 2022-03-30:
Are you able to login to the official Tesla App using that account? **Yes. I am able to login the official Tesla App with both my main account and my new dummy account without problem.**
Do you get the 403 when you entered your password, or directly when opening the URL? **With the latest version (with login_hint) the email address has already been populated in the pop up browser window. I got 403 after entering password and Enter. With the 33db201 (without login_hint) the only text box on that page is for the email address. I got 403 after entering email address and Enter (it did not even ask for the password, yet).**
Have you tried 3rd party token generation apps https://teslascope.com/help/generating-tokens? **I just tried the "Auth for Tesla" iOS app in that link you provided. I was able to obtain token with that app without issue.**

tdorssers, 2022-03-31:
Thanks, at least you can start using TeslaPy using the token you generated with the 3rd party token app. I believe that Tesla is refusing some user agents from accessing the v3 oauth URLs. Are you using any VPN software on your PC, which OS are you using? We could try to have selenium change the user agent string...

carrotcn, 2022-03-31:
No I do not have any VPN running on my PC. I am running windows 10 2004 (19041.1415).
The user agent I have is "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/100.0.4896.60 Safari/537.36". Would you mind sharing yours so that I can try replacing mine in selenium?

tdorssers, 2022-04-01:
I am able to reproduce the issue now, using Chrome controlled by selenium. I always test using a regular browser window, so your side note about that was helpful.

mandospirit, 2022-04-04:
Hi,
I'm running in the same issue here. Thought it was linked to my macos update, but it seems not after all.

I'm running cli.py and also getting the following error using --chrome arg:
`You don't have permission to access "http://auth.tesla.com/oauth2/v3/authorize?" on this server.`

Chrome (and chromedriver) is in version `100.0.4896.60`
Removing the login hint does not allow the connection.
I tried with Safari and it's the same behaviour.

You're saying you were able to reproduce it using Chrome controlled by Selenium, would you have a hint on how to bypass this error?

Thanks a lot.

carrotcn, 2022-04-05:
@mandospirit the latest version today should have fixed the issue on Chrome. The error is due to Tesla's detection mechanism of automated browser usage. At the moment the fix is able to circumvent the detection on Chrome and Edge Chromium, but not Fire Fox/Opera/Safari. You may download/clone the latest version and give it a try.

mandospirit, 2022-04-06:
Thanks a lot, will give it a try :)