tdorssers/TeslaPy #66: Custom cache load/save via ruamel.yaml is failing

philipsd6, 2022-04-02:
I'm trying to implement a `cache_loader` and `cache_dumper` with `ruamel.yaml`, but while I can use my functions directly, using them with the Tesla class fails.

(I realize that doing this in YAML vs the normal JSON doesn't add any value, I'm just doing it for experimentation and fun!)

My code:

```python
from teslapy import Tesla
from ruamel.yaml import YAML

def load_cache():
    yaml = YAML(typ="safe", pure=True)  # ensure we return a real dict()
    with open("cache.yml") as f:
        return yaml.load(f)

def save_cache(cache):
    yaml = YAML(typ="safe", pure=True)
    yaml.default_flow_style = False
    with open("cache.yml", "w") as f:
        yaml.dump(cache, f)

with Tesla(
    email="me@example.com", cache_loader=load_cache, cache_dumper=save_cache
) as tesla:
    tesla.fetch_token()
```
When I run this, I get this:
```python-traceback
File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts-5VgpqDjT-py3.10/lib/python3.10/site-packages/requests_oauthlib/oauth2_session.py", line 502, in request
    token = self.refresh_token(
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/teslapy/__init__.py", line 210, in refresh_token
    self._token_updater()  # Save new token
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/teslapy/__init__.py", line 274, in _token_updater
    self.cache_dumper(cache)
  File "/home/philipsd6/devel/teslapy/./tesla_test.py", line 46, in save_cache
    yaml.dump(cache, f)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/main.py", line 574, in dump
    return self.dump_all([data], stream, transform=transform)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/main.py", line 583, in dump_all
    self._context_manager.dump(data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/main.py", line 915, in dump
    self._yaml.representer.represent(data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 80, in represent
    node = self.represent_data(data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 103, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 321, in represent_dict
    return self.represent_mapping('tag:yaml.org,2002:map', data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 214, in represent_mapping
    node_value = self.represent_data(item_value)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 103, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 321, in represent_dict
    return self.represent_mapping('tag:yaml.org,2002:map', data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 214, in represent_mapping
    node_value = self.represent_data(item_value)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 113, in represent_data
    node = self.yaml_representers[None](self, data)
  File "/home/philipsd6/.cache/pypoetry/virtualenvs/tesla-scripts/lib/python3.10/site-packages/ruamel/yaml/representer.py", line 354, in represent_undefined
    raise RepresenterError(_F('cannot represent an object: {data!s}', data=data))
ruamel.yaml.representer.RepresenterError: cannot represent an object: {'access_token': '...redacted', 'refresh_token': '...redacted'}
```

But calling the functions manually works fine with no errors:
```python
In [3]: cache = load_cache()

In [4]: cache.keys()
Out[4]: dict_keys(['me@example.com'])

In [5]: save_cache(cache)
```

carrotcn, 2022-04-04:
@philipsd6 You might want to check [this](https://www.reddit.com/r/learnpython/comments/8no0d6/printing_a_nested_dictionary_as_a_yaml/) thread.

Specifically in your save_cache function if you instead use `yaml.dump(json.loads(json.dumps(cache)), f)` , that give you the desired output in cache.yml without problem. You might want to take extra steps when reading the cache as well, as mentioned in that linked article.

philipsd6, 2022-04-05:
I tried that and it worked, and then I went back to my way (because I want the round-trip YAML) and it's still working fine. I have no idea what was going on. But I'll chalk it up to PEBKAC

closed 2022-04-05 by philipsd6