tdorssers/TeslaPy #68: HTTP Status 400 when querying historic battery data

realdadfish, 2022-04-13:
I have a powerwall and try to get historic data via `tesla.battery_list()[0].get_history_data()`.

The built request for this however seems to be invalid. I get the following reported:

```
HTTPError: 400 Client Error: kind `savings` is not supported for url: https://owner-api.teslamotors.com/api/1/energy_sites/2653779/history?kind=savings&period=day&end_date=2022-04-14T00%3A13%3A08.000Z
```

israndy, 2022-04-14:
I wonder if this is related to my inability to get Charging History for my car. Even the Menu and GUI and CLI demo apps don't return correctly. Perhaps the API has changed with the new app update

tdorssers, 2022-04-17:
@realdadfish Please try to set kind to one of these: 'backup', 'energy', 'power', 'self_consumption', 'time_of_use_energy', 'time_of_use_self_consumption'
`tesla.battery_list()[0].get_history_data(kind='backup')`
This is a bug, as the default is set to 'savings' which is not valid. I don't own a Powerwall so I can't test this. Which value should I used as a default for this method?

@israndy This is not related. Are you using the latest commit or the version on pypi? It should work using the latest commit.

realdadfish, 2022-04-17:
So, 'time_of_use_energy' and 'time_of_use_self_consumption' both give me back an empty string ''. The other types give me back the following, without further arguments always data for the last two days including the current day:

### self_consumption

```
{
    "period": "day",
    "time_series": [
        {
            "timestamp": "2022-04-16T00:00:00+02:00",
            "solar": 39.04361328782189,
            "battery": 16.56170938579929
        },
        {
            "timestamp": "2022-04-17T00:00:00+02:00",
            "solar": 71.0134238360714,
            "battery": 28.9865761639286
        }
    ]
}
```

I don't know what these numbers should be, it's not energy (a Powerwall maxes out at 13.3kWh and I generate at most 35 to 40kWh of solar on good days).

### energy

```
{
    "serial_number": "1152100-13-E--JBL19206G3E097",
    "period": "day",
    "time_series": [
        {
            "timestamp": "2022-04-16T01:00:00+02:00",
            "solar_energy_exported": 38509.638404361904,
            "generator_energy_exported": 0,
            "grid_energy_imported": 22274.415243826807,
            "grid_services_energy_imported": 0,
            "grid_services_energy_exported": 0,
            "grid_energy_exported_from_solar": 2369.5945452470332,
            "grid_energy_exported_from_generator": 0,
            "grid_energy_exported_from_battery": 0,
            "battery_energy_exported": 8270,
            "battery_energy_imported_from_grid": 106.17325040325522,
            "battery_energy_imported_from_solar": 16643.826749596745,
            "battery_energy_imported_from_generator": 0,
            "consumer_energy_imported_from_grid": 22168.24199342355,
            "consumer_energy_imported_from_solar": 19496.217109518126,
            "consumer_energy_imported_from_battery": 8270,
            "consumer_energy_imported_from_generator": 0
        },
        {
            "timestamp": "2022-04-17T01:00:00+02:00",
            "solar_energy_exported": 28110.59808714874,
            "generator_energy_exported": 0,
            "grid_energy_imported": 78.99528539739549,
            "grid_services_energy_imported": 0,
            "grid_services_energy_exported": 0,
            "grid_energy_exported_from_solar": 2022.3109115064144,
            "grid_energy_exported_from_generator": 0,
            "grid_energy_exported_from_battery": 0,
            "battery_energy_exported": 5640,
            "battery_energy_imported_from_grid": 78.99528539739549,
            "battery_energy_imported_from_solar": 12271.004714602605,
            "battery_energy_imported_from_generator": 0,
            "consumer_energy_imported_from_grid": 0,
            "consumer_energy_imported_from_solar": 13817.282461039722,
            "consumer_energy_imported_from_battery": 5640,
            "consumer_energy_imported_from_generator": 0
        }
    ]
[... 73 lines cut ...]
        {
            "timestamp": "2022-01-25T12:39:05+01:00",
            "duration": 63083
        },
        {
            "timestamp": "2021-10-21T12:34:41+02:00",
            "duration": 83453
        },
        {
            "timestamp": "2021-07-06T19:37:32+02:00",
            "duration": 64695
        },
        {
            "timestamp": "2020-05-19T08:54:07+02:00",
            "duration": 261617
        }
    ],
    "events_count": 6,
    "total_events": 6
}
```

As expected, the single events where the Powerwall had to act as a backup source because of grid outages. All events are listed here for me, the duration is given here in milliseconds (we never had an outage that took longer than a few minutes).

So all in all I think the default should be `energy`, so users aren't thrown a list of several hundred records at when first trying out the API.

realdadfish, 2022-04-17:
Btw... the `get_calendar_history_data()` call also returns an empty list, because 'savings' is the default there as well, but at least it doesn't crash. If I give it a proper known 'kind' I get the same output like for the `get_history_data()`, but just for today (not two days).

tdorssers, 2022-04-17:
Thanks, I will change the default kind to energy. Shall I change the default kind of get_calendar_history_data() as well?

israndy, 2022-04-17:
> On Apr 17, 2022, at 1:28 AM, Tim Dorssers ***@***.***> wrote:
> 
> 
> 
> @israndy <https://github.com/israndy> This is not related. Are you using the latest commit or the version on pypi? It should work using the latest commit.
> 
Yes, TeslaPY appears to be returning just {}. Here is me starting afresh but getting the same results after downloading the .zip archive:

***@***.*** python3 % pip3 uninstall teslapy
Found existing installation: TeslaPy 2.4.0
Uninstalling TeslaPy-2.4.0:
  Would remove:
    /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/TeslaPy-2.4.0.dist-info/*
    /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/teslapy/*
Proceed (Y/n)? y
  Successfully uninstalled TeslaPy-2.4.0
***@***.*** python3 % pip3 install teslapy --no-cache-dir
Collecting teslapy
  Downloading TeslaPy-2.4.0-py2.py3-none-any.whl (33 kB)
Requirement already satisfied: requests-oauthlib in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from teslapy) (1.3.1)
Requirement already satisfied: requests>=2.4.2 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from teslapy) (2.27.1)
Requirement already satisfied: websocket-client>=0.59.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from teslapy) (1.3.2)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from requests>=2.4.2->teslapy) (1.26.9)
Requirement already satisfied: charset-normalizer~=2.0.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from requests>=2.4.2->teslapy) (2.0.12)
Requirement already satisfied: certifi>=2017.4.17 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from requests>=2.4.2->teslapy) (2021.10.8)
Requirement already satisfied: idna<4,>=2.5 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from requests>=2.4.2->teslapy) (3.3)
Requirement already satisfied: oauthlib>=3.0.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from requests-oauthlib->teslapy) (3.2.0)
Installing collected packages: teslapy
Successfully installed teslapy-2.4.0
***@***.*** python3 % cd teslapy-2.4.0
***@***.*** teslapy-2.4.0 % ls
Dockerfile		entrypoint.sh		setup.cfg
LICENSE			gui.py			setup.py
MANIFEST.in		media			teslapy
README.md		menu.py
cli.py			requirements.txt
***@***.*** teslapy-2.4.0 % python3 menu.py
Enter email: ***@***.***
Use browser to login. Page Not Found will be shown at success.
Enter URL after authentication: https://auth.tesla.com/void/callback?code=86bc97 <https://auth.tesla.com/void/callback?code=86bc97> {edited for privacy}
--------------------------------------------------------------------------------
ID Display name              VIN                       State                    
 0 Rosebud                   5YJ3E1EAXJFxxxxx         asleep                   
--------------------------------------------------------------------------------
Select vehicle: 0
--------------------------------------------------------------------------------
VIN decode: Tesla Motors, Inc., Tesla Model 3 {edited for brevity}
--------------------------------------------------------------------------------
Wake up vehicle to use remote functions/telemetry
--------------------------------------------------------------------------------
 1 Refresh                 2 Charging history        3 Wake up                
 4 Nearby charging sites   5 Honk horn               6 Flash lights           
 7 Lock/unlock             8 Climate on/off          9 Set temperature        
10 Actuate frunk/trunk    11 Remote start drive     12 Set charge limit       
13 Open/close charge port 14 Start/stop charge      15 Seat heater request    
16 Toggle media playback  17 Window control         18 Max defrost            
19 Set charging amps      
--------------------------------------------------------------------------------
Choice (0 to quit): 2
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 336, in <module>
    main()
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 319, in main
    menu(vehicles[idx])
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 204, in menu
    show_charging_history(vehicle.get_charge_history())
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 143, in show_charging_history
    print(data['screen_title'])
TypeError: string indices must be integers
[... 37 lines cut ...]
Car Type: model3                       Exterior Color: RedMulticoat
Wheel Type: StilettoRefresh19          Spoiler Type: None
Roof Color: RoofColorGlass             Charge Port Type: US
--------------------------------------------------------------------------------
 1 Refresh                 2 Charging history        3 Wake up                
 4 Nearby charging sites   5 Honk horn               6 Flash lights           
 7 Lock/unlock             8 Climate on/off          9 Set temperature        
10 Actuate frunk/trunk    11 Remote start drive     12 Set charge limit       
13 Open/close charge port 14 Start/stop charge      15 Seat heater request    
[... cut ...]
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 336, in <module>
    main()
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 319, in main
    menu(vehicles[idx])
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 204, in menu
    show_charging_history(vehicle.get_charge_history())
  File "/Users/israndy/Documents/Xcode/Python3/TeslaPy-2.4.0/menu.py", line 143, in show_charging_history
    print(data['screen_title'])
TypeError: string indices must be integers
***@***.*** teslapy-2.4.0 %

realdadfish, 2022-04-17:
Yes, that would make sense.

tdorssers, 2022-04-18:
@israndy The latest commit is not a zip file. You should pull this repo to get the latest commit (or simply download __init__.py). I will package version 2.5.0 shortly which includes a fix.

tdorssers, 2022-04-18:
@realdadfish commit e88ec08 includes the fixed kinds.

commit e88ec083 (2022-04-18): Added #65, fixed #68 and added #71
closed 2022-05-05 by tdorssers