tdorssers/TeslaPy #70: Battery percentage_charged is wrong

realdadfish, 2022-04-16:
I'm not exactly sure what `percentage_charged`  means, but dividing `4731 Wh / 13774 Wh` gives me 34% charge state, which is also displayed in the app. The `3.868...` is wrong in my opinion.

```
{
    ...
    "energy_left": 4731.894736842106,
    "total_pack_energy": 13774,
    "percentage_charged": 3.8685643327920523,
    ...
}
```

tdorssers, 2022-04-17:
Teslapy doesn't set these values, they are set by Tesla's backend. I'm also not sure what all values mean, as I don't own a powerwall.

realdadfish, 2022-04-17:
Hrm... maybe it's the last charge amount, but I have no idea. At 100% I get the following:

```
    "energy_left": 13775,
    "total_pack_energy": 13775,
    "percentage_charged": 88.20899799407775,
```

Will close this then.

closed 2022-04-17 by realdadfish