tdorssers/TeslaPy #72: Using refresh token to authenticate does not work

realdadfish, 2022-04-16:
I generate a specific refresh token like this:

```
import teslapy
import sys

with teslapy.Tesla(input('Enter Tesla Login E-Mail: '), cache_loader=lambda: {}, cache_dumper=lambda x: None) as tesla:
    if not tesla.authorized:
        print('Use browser to login. Page Not Found will be shown at success.')
        print('Open this URL: ' + tesla.authorization_url())
        try:
            token = tesla.fetch_token(authorization_response=input('Enter URL after authentication: '))
            print(f"Refresh Token: {token['refresh_token']}")
        except Exception as e:
            print(f"Authorisation failed: {e}")
            sys.exit(1)
    else:
        print(f"Refresh Token: {tesla.token['refresh_token']}")
```

Note how I disable `cache_loader` and `cache_dumper` here, so TeslaPy does not cache a successful authentication in between.

Now I use the returned `refresh_token` in another script to authenticate:

```
with teslapy.Tesla("<entered-email>") as tesla:
            if not self.tesla.authorized:
                try:
                    self.tesla.refresh_token(refresh_token = "<my-copied-refresh-token>")
                except Exception as e:
                    raise Exception("Refreshing the access token failed; is the refresh_token still valid?") from e
```

and this gives me the following error:

```
  File "/usr/local/lib/python3.9/site-packages/teslapy/__init__.py", line 209, in refresh_token
    super(Tesla, self).refresh_token(token_url, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests_oauthlib/oauth2_session.py", line 452, in refresh_token
    self.token = self._client.parse_request_body_response(r.text, scope=self.scope)
  File "/usr/local/lib/python3.9/site-packages/oauthlib/oauth2/rfc6749/clients/base.py", line 448, in parse_request_body_response
    self.token = parse_token_response(body, scope=scope)
  File "/usr/local/lib/python3.9/site-packages/oauthlib/oauth2/rfc6749/parameters.py", line 441, in parse_token_response
    validate_token_parameters(params)
  File "/usr/local/lib/python3.9/site-packages/oauthlib/oauth2/rfc6749/parameters.py", line 448, in validate_token_parameters
    raise_from_error(params.get('error'), params)
  File "/usr/local/lib/python3.9/site-packages/oauthlib/oauth2/rfc6749/errors.py", line 399, in raise_from_error
    raise cls(**kwargs)
oauthlib.oauth2.rfc6749.errors.ServerError: (server_error) Internal server error
```

What am I doing wrong?

tdorssers, 2022-04-17:
I tested your code snippets and I don't get a ServerError exception. Which version of teslapy are you using? I used the latest commit.

realdadfish, 2022-04-17:
Ok, my fault, I had a stupid programming error, it is working indeed perfectly fine. Thank you!

closed 2022-04-17 by realdadfish