tdorssers/TeslaPy #74: Setting passenger temperature results in  lowest setting.

johnypony3, 2022-04-23:
I'd like to thank you for this body of work. Very well crafted!

Setting the drivers temperature works as expected.

'''json
{"name":"change_climate_temperature_setting","data":{"driver_temp":20.555555555555554},"result":true}

{"name":"change_climate_temperature_setting","data":{"passenger_temp":20.555555555555554},"result":true}
'''

[image]

tdorssers, 2022-04-23:
You need to pass both driver_temp and passenger_temp. Also both values should be equal.

cli.py -e elon@tesla.com -c CHANGE_CLIMATE_TEMPERATURE_SETTING -k driver_temp=20 -k passenger_temp=20

johnypony3, 2022-04-23:
that definitely worked, any insight into why they have to be equal?

closed 2022-04-23 by johnypony3