tdorssers/TeslaPy #76: Cannot login

ben1628, 2022-04-30:
I'm trying to run the menu.py and get this error after I enter my email address

Enter email: test@outlook.com
[pywebview] loaded event is deprecated and will be removed in 4.0. Use events.loaded instead
2022-04-30 15:41:20,351 - pywebview - WARNING - loaded event is deprecated and will be removed in 4.0. Use events.loaded instead
Traceback (most recent call last):
  File "c:\SourceVector\TeslaPy-master\TeslaPy-master\menu.py", line 345, in <module>
    main()
  File "c:\SourceVector\TeslaPy-master\TeslaPy-master\menu.py", line 311, in main
    vehicles = tesla.vehicle_list()
  File "c:\SourceVector\TeslaPy-master\TeslaPy-master\teslapy\__init__.py", line 369, in vehicle_list
    return [Vehicle(v, self) for v in self.api('VEHICLE_LIST')['response']]
  File "c:\SourceVector\TeslaPy-master\TeslaPy-master\teslapy\__init__.py", line 354, in api
    self.fetch_token()
  File "c:\SourceVector\TeslaPy-master\TeslaPy-master\teslapy\__init__.py", line 221, in fetch_token
    kwargs['authorization_response'] = self.authenticator(url)
  File "c:\SourceVector\TeslaPy-master\TeslaPy-master\menu.py", line 280, in custom_auth
    window.loaded += on_loaded
AttributeError: can't set attribute
PS C:\SourceVector> 

Any ideas?

I'm using Python 3.8.8

If I run this in debug mode, I have this error

Exception has occurred: TypeError
vars() argument must have __dict__ attribute
  File "C:\SourceVector\TeslaPy-master\TeslaPy-master\menu.py", line 25, in <module>
    raw_input = vars(__builtins__).get('raw_input', input)  # Py2/3 compatibility

Thanks

88gts, 2022-05-02:
cant give you any tips but will say that my system started experiencing login issues around 4/30 evening PST.  tried blowing away cache.json and re-authing but to no avail.

tdorssers, 2022-05-02:
Please try commit 54c8abe as it contains a fix to support both pywebview 3.6+ and older versions.

88gts, 2022-05-02:
no issues seen here, thanks!

ben1628, 2022-05-04:
It works.  Thanks a lot

closed 2022-05-04 by ben1628