tdorssers/TeslaPy #77: ChargerPhases odd behaviour

dmaunder, 2022-05-19:
I am finding an odd result with the ChargerPhases value.

If the charger is single phase, such as my home charger, it returns the correct value of 1

If the charger is 3 phase (and I believe it is, as it shows 16A 3 phase in the car and is charging at 11kw which is what it would if it was 3 phase, it returns a value of 2. I have this at 2 different locations.

The code I am using is

            ChargerPhases = int(car.get_vehicle_data()['charge_state']['charger_phases'])

tdorssers, 2022-05-19:
This is actually expected behavior. You should not cast to int, because 'charger_phases' can be `None`. 1 means single phase and 2 means three phases. These values are set by the server and are undocumented.

dmaunder, 2022-05-20:
Thanks Tim.

I was doing the cast only if charging_state was set to 'Charging', so it shouldnt be None in that case, plus it is in an exception handler anyway. However The 1 means single phase and 2 means 3 phase is interesting. What would it do in the USA where they commonly use 2 phase?

tdorssers, 2022-06-04:
I don't think that Tesla supports 2 phase. You need a Scott-T transformer in such a case I think.

closed 2022-07-03 by tdorssers