tdorssers/TeslaPy #78: Commands Hanging On New Ubuntu Installation

Haselmaier, 2022-05-25:
I've been using Teslapy for about 6 months.  Awesome.  I use it to adjust charging based on solar panel output.

I recently built up an Ubuntu machine.  The code hangs when executing a command.  See the attached output when I stop the code.  This code runs on other machines (e.g. Windows) but won't run on this Ubuntu machine.

All the packages identified in requirements.txt seem to be up to minimum versions required.

Any guidance on what I need to do to rectify this?

Thanks.

Jim Haselmaier
[image]

tdorssers, 2022-05-25:
Are you using different versions of TeslaPy on your machines? This requests exception probably occurs because the Tesla instance is accessed outside of the context manager. Recent versions of TeslaPy destroy the connection adapter when the context manager exits.

Haselmaier, 2022-05-25:
Thanks Tim.

One of my other machines was running 2.2.0 and the Ubuntu machine is running 2.5.0.  I moved the machine that was working well to 2.5.0 and confirmed it is behaving the same way as the error messages I posted in my first post.

I'm not following your comment (sorry - I'm not too technical) about recent versions of TeslaPy and destroying the connection adapter.  Any guidance on what is different in 2.5.0 that I need to adapt to to get my code to work?

Thanks!

tdorssers, 2022-05-26:
Maybe these two examples help. A context manager automatically closes the requests session. TeslaPy 2.5.0 also closes the connection adapter. This has been done because the requests module doesn't reconnect on network failures once it has been closed. So for long running scripts TeslaPy 2.5.0 is the recommended version.

With context manager:
```python
with teslapy.Tesla('elon@tesla.com') as tesla:
	vehicles = tesla.vehicle_list()
	print(vehicles[0])
print(vehicles[0])  # Causes exception
```
Without context manager:
```python
tesla = teslapy.Tesla('elon@tesla.com')
vehicles = tesla.vehicle_list()
print(vehicles[0])
tesla.close()
print(vehicles[0])  # Causes exception
```

closed 2022-07-03 by tdorssers