tdorssers/TeslaPy #85: Python error introduced sometime after v2.4.0

ralequi, 2022-07-08:
Hello,

I'm using Fedora Linux with python 3.10.5.

With version v2.4.0 there is no issue, but with higher versions I get the following error:

```
(.env) [rleira@jenova TeslaPy]$ python gui.py 
Traceback (most recent call last):
  File "/home/rleira/git/TeslaPy/gui.py", line 1323, in <module>
    app = App()
  File "/home/rleira/git/TeslaPy/gui.py", line 617, in __init__
    self.browser = IntVar(0)
  File "/usr/lib64/python3.10/tkinter/__init__.py", line 564, in __init__
    Variable.__init__(self, master, value, name)
  File "/usr/lib64/python3.10/tkinter/__init__.py", line 372, in __init__
    self._root = master._root()
AttributeError: 'int' object has no attribute '_root'
```

Of course i've tried to run (many times in fact) `pip install --upgrade -r requirements.txt ; pip install --upgrade .` on master, v2.6 and v2.5 but without any working result.

ralequi, 2022-07-08:
I've performed a git bisect. 
The result was the following:

```
08e8cd33b2bc9c82c161e108a67abc134781501e is the first bad commit
commit 08e8cd33b2bc9c82c161e108a67abc134781501e
Author: tdorssers <tim.dorssers@xs4all.nl>
Date:   Sat Apr 9 14:21:12 2022 +0200

    Removed Firefox and Safari support

 README.md | 14 +++-----------
 cli.py    | 26 +++++++++++++-------------
 gui.py    | 22 ++++++++++------------
 menu.py   | 26 +++++++++++++-------------
 4 files changed, 39 insertions(+), 49 deletions(-)

```

ralequi, 2022-07-08:
Possible fix:

I've tried to change:

```python
self.browser = IntVar(0)
```

to

```python
self.browser = IntVar()
```

And it seems it fixes the gui and works again. Don't know if it may cause side effects/errors in other environments

ralequi, 2022-07-08:
I've tried to change:

```python
self.browser = IntVar(0)
```

to

```python
self.browser = IntVar()
```

And it seems it fixes some parts. For example, the gui stop crashing, it gets the vehicle main data (option codes and vehicle color) but nothing else.
Probably is not enough but an starting point

closed 2022-07-08 by tdorssers

tdorssers, 2022-07-08:
Thanks for this find. Actually the default value is 0 so it can be omitted as well.