tdorssers/TeslaPy #100: get_charge_history/VEHICLE_CHARGE_HISTORY has timezone GMT-07

jankratochvil, 2022-09-02:
Hello, the charging history shows different values than what the (Android) Tesla app shows me. That is apparently because the app shows bars/days according to my timezone midnight (CEST=GMT+02) but TeslaPy receives data with timestamps of GMT-07 (such as 1661410800) midnight.
Do you have some recorded data from the Tesla app? I expect the app sends some extra header / JSON data specifying local timezone.

KNiepok, 2022-09-15:
@jankratochvil Hi, unfortunately I don't know answer to your question, but I'm currently trying to use this endpoint and all I receive from it is an empty response along with a 200OK code:
```
GET https://owner-api.teslamotors.com/api/1/vehicles/{$id}/charge_history
X-Tesla-User-Agent: TeslaApp/4.12.1
User-Agent: River
Authorization: Bearer {$token}
Host: owner-api.teslamotors.com
```

```
{
    "response": "{}\n"
}
```
Can you explain how did you manage to get results from that endpoint?

jankratochvil, 2022-09-15:
You need to replace the {$id} there, in my case:
https://owner-api.teslamotors.com/api/1/vehicles/29342XXX0953/charge_history
XXX are 3 digits.

jankratochvil, 2022-09-15:
Do you get the graphical output from `gui.py`? I do. Then I added some debug output to TeslaPy files as I was porting it to: https://github.com/stevieb9/tesla-vehicle

jan-gogo, 2022-09-15:
@jankratochvil I have the same problem the `gui.py` doesn't render any data on charge history screen, only empty diagram.  The response code is 200 and json is empty.

tdorssers, 2022-09-16:
I have no output, just "response": "{}\n" because I am leasing my car. I am not the owner, just the driver, so this call doesn't work for me. I also have no recorded data from the Tesla App to see if there are any parameters that can set the time zone for the call.
@jankratochvil can you share the output of the call, so I can validate the graphical output?

jankratochvil, 2022-09-17:
The Android app shows it correctly as its midnight is at real midnight. But gui.py gets it shifted.
gui.py.txt contains for example:
```
                                                           'timestamp': {'seconds': 1660806000}},
```
Which is Arizona U.S. timezone.
```
$ TZ=US/Arizona    perl -MPOSIX -e 'print ctime 1660806000'
Thu Aug 18 00:00:00 2022
$ TZ=GMT           perl -MPOSIX -e 'print ctime 1660806000'
Thu Aug 18 07:00:00 2022
$ TZ=Europe/Prague perl -MPOSIX -e 'print ctime 1660806000'
Thu Aug 18 09:00:00 2022
```
It looks as it is really split by such U.S. timezone midnight as normally I charge in the morning (before 9am, from 50% to 90%) before driving and in the evening (to restore SoC back to 50%) which Tesla Android app correctly displays as a single big bar. But gui.py does show such day as two charging days.

[image]
[image]
[gui.py.txt](https://github.com/tdorssers/TeslaPy/files/9592477/gui.py.txt)