tdorssers/TeslaPy #101: Is it possible to authenticate without using a browser at all - only with console?

philhzss, 2022-09-10:
Hi,

Sorry if stupid question. I wrote a little Tesla climate automator app a few years ago for myself, it's been dead ever since Tesla changed the login procedure earlier this year. I'm trying to fix it now, this script of mine runs on a headless raspberry pi, so I can't have browser windows opening to authenticate... I want it to run all by itself once setup.

I tried using selenium since you write it can automate browser interaction, but I don't understand, where am I suppose to pass my Tesla password. With this example:

```
import teslapy
from selenium import webdriver
from selenium.webdriver.support import expected_conditions as EC
from selenium.webdriver.support.ui import WebDriverWait

def custom_auth(url):
    options = webdriver.ChromeOptions()
    options.add_argument("--disable-blink-features=AutomationControlled")
    with webdriver.Chrome(chrome_options=options) as browser:
        browser.get(url)
        WebDriverWait(browser, 300).until(EC.url_contains('void/callback'))
        return browser.current_url

with teslapy.Tesla('elon@tesla.com', authenticator=custom_auth) as tesla:
    tesla.fetch_token()
```

I suppose it's designed to open a browser window, and then I would enter my password there? I can't get it to work (most likely my fault), but is there a way to skip all that and do everything with the terminal, to make it headless?

Thank you!

gbizeau, 2022-09-10:
Go here to get your "refresh Token'

https://tesla-info.com/tesla-token.php

Once you have that you can run your script on the CLI and input that value, once your script has this and runs consistently, 
it will not ask again.

Use this bit of code for Auth on your PI

    with teslapy.Tesla('elon@tesla.com') as tesla:
        if not tesla.authorized:
            tesla.refresh_token(refresh_token=input('Enter SSO refresh token: '))

philhzss, 2022-09-10:
Oh, that was deceptively simple. I had the TeslaTokens app on my phone, but I was under the (very wrong) impression that a "refresh token" is just used to "refresh" an already created access token... as-in, I thought I had to use my password to generate an access token, and only then would I be able to refresh it.

Your example worked perfectly. 

Thank you!!!

closed 2022-09-10 by philhzss