tdorssers/TeslaPy #102: Upgrade of Tesla App breaks API calls

gbizeau, 2022-09-17:
within 30 minutes of upgrading my iPhone App to 4.13.1-1312 My App started displaying SSL errros

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 1042, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 414, in connect
    self.sock = ssl_wrap_socket(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/ssl_.py", line 449, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/ssl_.py", line 493, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.10/ssl.py", line 512, in wrap_socket
    return self.sslsocket_class._create(
  File "/usr/lib/python3.10/ssl.py", line 1070, in _create
    self.do_handshake()
  File "/usr/lib/python3.10/ssl.py", line 1341, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLZeroReturnError: TLS/SSL connection has been closed (EOF) (_ssl.c:997)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='owner-api.teslamotors.com', port=443): Max retries exceeded with url: /api/1/vehicles (Caused by SSLError(SSLZeroReturnError(6, 'TLS/SSL connection has been closed (EOF) (_ssl.c:997)')))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/./teslatracker.py", line 41, in <module>
    vehicles = tesla.vehicle_list()
  File "/usr/local/lib/python3.10/dist-packages/teslapy/__init__.py", line 369, in vehicle_list
    return [Vehicle(v, self) for v in self.api('VEHICLE_LIST')['response']]
  File "/usr/local/lib/python3.10/dist-packages/teslapy/__init__.py", line 363, in api
    return self.request(endpoint['TYPE'], uri, serialize,
  File "/usr/local/lib/python3.10/dist-packages/teslapy/__init__.py", line 147, in request
    response = super(Tesla, self).request(method, url, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests_oauthlib/oauth2_session.py", line 521, in request
    return super(OAuth2Session, self).request(
  File "/usr/local/lib/python3.10/dist-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/adapters.py", line 563, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='owner-api.teslamotors.com', port=443): Max retries exceeded with url: /api/1/vehicles (Caused by SSLError(SSLZeroReturnError(6, 'TLS/SSL connection has been closed (EOF) (_ssl.c:997)')))
```

jeremy-compostella, 2022-09-17:
I am also running into similar issue. Did  you find any workaround ?

jeremy-compostella, 2022-09-17:
I think that's most likely an issue on the server side. Even the app show 503 Server Maintenance sometimes.

mababio, 2022-09-18:
Experiencing same issue. Do you know how long the bad response last? I was thinking about building an exception handler that would retry.

gbizeau, 2022-09-18:
Mine seems to be working again now. I have experienced the odd 503 before, but never to this extent and the timing was of major / concerning coincidence.

mababio, 2022-09-18:
Have an automation using teslapy ... it will be running for  at least 5 hours. So let's see how things pan out!! Fingers crossed

mababio, 2022-09-19:
All good! Teslapy has been running for more than 5 hours

closed 2022-12-11 by tdorssers