tdorssers/TeslaPy #103: looks like charge_state has moved

doubledrat, 2022-09-24:
for me at least it's now under ['data']

```
        "data": {
            "charge_state": {
                "battery_heater_on": false,
                "battery_level": 82,
    
```

which breaks last_seen()

looks like gui_settings is also under there now, which will break a few other things

tdorssers, 2022-09-24:
Actually the response of get_latest_vehicle_data() has changed:
```
{
    "pb_data": "EiIKAg...",
    "data": {
        "charge_state": {
            "battery_heater_on": false,
etc.
```
I am going to make a fix for this.

tdorssers, 2022-09-24:
"CACHED_PROTO_VEHICLE_DATA" has been renamed to "LATEST_VEHICLE_DATA" in official app version 4.12 and it has been removed in app version 4.13. I am guessing that "VEHICLE_DATA" is now returning cached data when the vehicle is asleep.

kaistian, 2022-12-19:
I tried the example in the README.md with `2.7.0` and `vehicles[0]['charge_state']['battery_level']` gives me `0`.
I needet to use `vehicles[0].get_vehicle_data()['charge_state']['usable_battery_level']` to get the correct SoC.

tdorssers, 2023-05-21:
Should be fixed in commit 87fd806

closed 2023-05-21 by tdorssers