tdorssers/TeslaPy #111: May be an issue with CHANGE_CLIMATE_TEMPERATURE_SETTING command

mababio, 2022-10-22:
I have been running this code for some time. It sets cabin temperature:

``` with teslapy.Tesla(settings['production']['teslapy']['email']) as tesla:
            vehicles = tesla.vehicle_list()
            vehicles[0].sync_wake_up()
            professor = vehicles[0]
            tesla_data_climate = professor.api('VEHICLE_DATA')['response']['climate_state']
            if not tesla_data_climate['is_climate_on']:
                professor.command('CLIMATE_ON')
            professor.command('CHANGE_CLIMATE_TEMPERATURE_SETTING', driver_temp=temp, passenger_temp=temp)```

for some reason it's setting the temp to `Lo`  I set temp to 22.0

Could someone check they're experiencing the same behavior

tdorssers, 2022-10-30:
I just checked this but no it works as expected for me using car sw 2022.36.5

mababio, 2022-11-10:
Strange, works now. thanks

closed 2022-11-10 by mababio