tdorssers/TeslaPy #116: 'drive_state' Data Sometimes Not Provide Following 'get_vehicle_data'?

Haselmaier, 2023-01-01:
It appears that sometimes the 'drive_state' dataset is sometimes not provided with a get_vehicle_data call.

Could that be possible.....or maybe there's an issue on my end?  My code will run for quite a while - checking some fields in that drive_state section.  Then I'll get the following and aborts:

Traceback (most recent call last):
  File "/home/pi/Desktop/Tesla/Suncatcherv5.3.py", line 292, in <module>
    at_home, tpi,odometer_short = tesla_plugged_in()
  File "/home/pi/Desktop/Tesla/Suncatcherv5.3.py", line 16, in tesla_plugged_in
    lat_raw = cardata['drive_state']['latitude']
  File "/home/pi/.local/lib/python3.9/site-packages/teslapy/__init__.py", line 610, in __missing__

The errors/exceptions are pretty lengthy.....but this is the beginning.  I've created a test program to only get the data and print it out.  Sometimes that section is there....and sometimes it's not.

Thoughts?

Thanks!

tdorssers, 2023-01-01:
This happens when the vehicle is asleep, because the owner API returns cached data when you make this call. If you call sync_wake_up() before get_vehicle_data(), then this should not happen.

Haselmaier, 2023-01-01:
Thank you Tim!

Have you seen this happen even when a sync_wake_up() is done?

I'll see how things do today.  But I seemed to have an occurrence of this yesterday when the car had been woken up.

with teslapy.Tesla('someone@gmail.com') as tesla:
        vehicles = tesla.vehicle_list()
        vehicles[0].sync_wake_up()
        cardata = vehicles[0].get_vehicle_data()
lat_raw = cardata['drive_state']['latitude']

Thank you again.

Jim

Haselmaier, 2023-01-02:
Additional info:

My code crashed this morning....after running for 18+ hours.  The overall context:  This program adjusts car charging based on solar panel output and home consumption.  The design case is for it to run 24x7....and if car is below certain charge at night charge at night just to a basic minimum we need.  My point:  It's pinging the car periodically for status.  During daylight it's pinging it about every 10 minutes.  It had been running quite a long time - with many successful car polls - when this had occurred.

Any ideas?

Thanks so much!

[SCError.txt](https://github.com/tdorssers/TeslaPy/files/10333140/SCError.txt)

tdorssers, 2023-01-03:
This exception is common if you access the TeslaPy object outside of a context handler. Here is an example without context handler: https://github.com/tdorssers/TeslaPy#alternative
It is also possible that the connection has dropped to the server. You can enable automatic retries like this: https://github.com/tdorssers/TeslaPy#robustness
Hope this helps.

jeffh0821, 2023-02-13:
Just to add on to this a bit...

I have also observed some inconsistent behavior with the return of `'drive_state'` data after the call of the `get_vehicle_data()` method on a `Vehicle` object. I concur with Tim that I think it has to do with the vehicle being `'asleep`', but I can also verify that the top-level `'state'` is `'online'`. In fact, I have not seen a state other than `'online'` since I've started working on my project.

I really do not think this is an issue with the library - more of an API nuance. Is there any way for us to reliably establish the case in which a `get_vehicle_data()` call will NOT return `'drive_state'` data?