tdorssers/TeslaPy #117: sync_wake_up Works Fine With One Car But (Frequently) Times Out On Another

Haselmaier, 2023-01-01:
One car (mine - 2022 Model Y) tends to handle the sync_wake_up() call just fine.  Another Tesla owner who is running my code (another Model Y - pre-2022) and it times out frequently on the sync_wake_up() call.

I believe the timeout can be changed - although I haven't looked into it closely or how to use it.

I'm really more interested in understanding why the difference in behavior.  Are there car differences that can cause this?  Maybe usage differences (my friend's car doesn't get a ton of use so....for example.....is the longer it's asleep does it take longer to wake it up?).

I'm mostly just curious as to why....although any guidance from someone who has messed with changing timeout would also be of interest.

Thanks.

tdorssers, 2023-01-03:
The exception you are getting is passed from the owner API to TeslaPy, so there is nothing that TeslaPy can do about this: the car has no connection to Hermes, which is a Tesla server or service. This can be caused by bad Wifi/cellular reception. Maybe you can simply retry your connection after 10 minutes or so and see what happens?

gbizeau, 2023-01-03:
Try / Except blocks are your friends, you can't always get data from
the car, you have to program for that. :)

closed 2023-05-21 by tdorssers