tdorssers/TeslaPy #118: TypeError: unsupported format string passed to NoneType.__format__

cycl0ne, 2023-01-04:
HI,

treid your little programs. the menu.py threw me this:

Traceback (most recent call last):
  File "/home/pos/python/./menu.py", line 347, in <module>
    main()
  File "/home/pos/python/./menu.py", line 326, in main
    menu(vehicles[idx])
  File "/home/pos/python/./menu.py", line 180, in menu
    show_vehicle_data(vehicle.get_vehicle_data())
  File "/home/pos/python/./menu.py", line 75, in show_vehicle_data
    print(fmt.format(door.get(ve['ft']), door.get(ve['rt'])))
TypeError: unsupported format string passed to NoneType.__format__

cycl0ne, 2023-01-04:
Ok seems there are more errors coming in.. on my second vehicle (MYC) : 
VIN decode: Tesla Motors, Inc., Tesla Model Y, Class D MPV / 5 Dr / LHD, Type 2 manual seatbelts (FR, SR*2, TR*2) with front airbags, PODS, side inflatable restraints, knee airbags (FR), Electric (NMC), Unknown, 2022, Shanghai, China
Traceback (most recent call last):
  File "/home/pos/python/./menu.py", line 347, in <module>
    main()
  File "/home/pos/python/./menu.py", line 324, in main
    print('Option codes:', ', '.join(vehicles[idx].option_code_list()))
  File "/home/poos/.local/lib/python3.9/site-packages/teslapy/__init__.py", line 540, in option_code_list
    for code in self['option_codes'].split(',')]))
AttributeError: 'NoneType' object has no attribute 'split'

cycl0ne, 2023-01-05:
Oh and another Question, while im trying to understand how to use the class:
```
import teslapy
tesla = teslapy.Tesla('claus@poweros.de')
if not tesla.authorized:
    print('Use browser to login. Page Not Found will be shown at success.')
    print('Open this URL: ' + tesla.authorization_url())
    tesla.fetch_token(authorization_response=input('Enter URL after authentication: '))
vehicles = tesla.vehicle_list()
vehicles[0].sync_wake_up()
print(vehicles[0]['charge_state'])
tesla.close()
```
Allways dumps me everything 0??
{
    "battery_heater_on": false,
    "battery_level": 0,
    "battery_range": 0.0,
    "charge_current_request": 0,
    "charge_current_request_max": 0,
    "charge_enable_request": false,
    "charge_energy_added": 0.0,
    "charge_limit_soc": 0,
    "charge_limit_soc_max": 0,
    "charge_limit_soc_min": 0,
    "charge_limit_soc_std": 0,
    "charge_miles_added_ideal": 0.0,
    "charge_miles_added_rated": 0.0,
    "charge_port_cold_weather_mode": false,
    "charge_port_color": "Off",
    "charge_port_door_open": false,
    "charge_port_latch": "Engaged",
    "charge_rate": 0,
    "charge_to_max_range": false,
    "charger_actual_current": 0,
    "charger_phases": 0,
    "charger_pilot_current": 0,
    "charger_power": 0,
    "charger_voltage": 0,
    "charge_amps": 0,
    "charging_state": "Disconnected",
    "conn_charge_cable": "IEC",
    "est_battery_range": 0.0,
    "fast_charger_brand": "SNA",
    "fast_charger_present": false,
    "fast_charger_type": "Supercharger",
    "ideal_battery_range": 0.0,
    "managed_charging_active": false,
    "managed_charging_start_time": 0,
    "managed_charging_user_canceled": false,
    "max_range_charge_counter": 0,
    "minutes_to_full_charge": 0,
    "time_to_full_charge": 0.0,
    "not_enough_power_to_heat": false,
    "off_peak_charging_enabled": false,
    "off_peak_charging_times": "all_week",
    "off_peak_hours_end_time": 0,
    "preconditioning_enabled": false,
    "preconditioning_times": "all_week",
    "scheduled_charging_mode": "Off",
    "scheduled_charging_pending": false,
    "scheduled_charging_start_time": 0,
    "scheduled_charging_start_time_app": 0,
    "scheduled_charging_start_time_minutes": 0,
    "scheduled_departure_time_minutes": 0,
    "supercharger_session_trip_planner": false,
    "timestamp": 1672913759730,
    "trip_charging": false,
    "usable_battery_level": 0,
    "user_charge_enable_request": false
}

tdorssers, 2023-01-08:
I can't reproduce these errors and I have never seen all zero's returned by the API. Does cli.py work for you? Can you do python cli.py -e claus@poweros.de -w -g?

cycl0ne, 2023-01-09:
Ok, the cli.py works smoothless.  all values are there. I tried this again after the successful cli.py

```
import teslapy
tesla = teslapy.Tesla('claus@poweros.de')
if not tesla.authorized:
    print('Use browser to login. Page Not Found will be shown at success.')
    print('Open this URL: ' + tesla.authorization_url())
    tesla.fetch_token(authorization_response=input('Enter URL after authentication: '))
vehicles = tesla.vehicle_list()
vehicles[1].sync_wake_up()
print(vehicles[1]['charge_state'])
tesla.close()
```

Everything is 0... 
    "battery_heater_on": false,
    "battery_level": 0,
    "battery_range": 0.0,
    "charge_current_request": 0,
    "charge_current_request_max": 0,
    "charge_enable_request": false,
    "charge_energy_added": 0.0,
    "charge_limit_soc": 0,
    "charge_limit_soc_max": 0,
    "charge_limit_soc_min": 0,
    "charge_limit_soc_std": 0,

If i start cli.py again.. all values are there.. hmm... where is the bug in this short code? :-)

cycl0ne, 2023-01-09:
Ok maybe found something. in the Example you write:
-> automatically calls get_latest_vehicle_data() when a key is not found.

This isnt done. My small script works if i change to this:
```
vehicles = tesla.vehicle_list()
vehicles[1].sync_wake_up()
vehicles[1].get_vehicle_data()
print(vehicles[1]['charge_state'])
tesla.close()
```
Cheers
C.

gbizeau, 2023-01-09:
Do you have two Teslas? Because you're gathering data from vehicles[1],
instead of vehicles[0]?

cycl0ne, 2023-01-09:
Yes correct. Got two Teslas and allways trying 0 and 1 just to check if both behave same.

tdorssers, 2023-01-14:
It seems that USER_ACCOUNT_GET_DETAILS is no longer available.

tdorssers, 2023-04-27:
Latest commit should fix the option list TypeError, as vehicle options are now None.