tdorssers/TeslaPy #124: last_seen() raises

paulhargreaves, 2023-04-26:
Simply calling:

`vehicles = self._tesla.vehicle_list()
print(vehicles[0].last_seen())`

causes:

`Traceback (most recent call last):
  File "/home/me/solar.py", line 32, in <module>
    main()
  File "/home/me/solar.py", line 21, in main
    car = Tesla()
  File "/home/me/classes/tesla.py", line 41, in __init__
    print(vehicles[0].last_seen())
  File "/home/me/mqtt/lib/python3.10/site-packages/teslapy/__init__.py", line 644, in last_seen
    diff = time.time() - self['charge_state']['timestamp'] / 1000
  File "/home/me/mqtt/lib/python3.10/site-packages/teslapy/__init__.py", line 610, in __missing__
    if key not in self.get_latest_vehicle_data():
  File "/home/me/mqtt/lib/python3.10/site-packages/teslapy/__init__.py", line 552, in get_latest_vehicle_data
    response = self.api('CACHED_PROTO_VEHICLE_DATA')['response']
  File "/home/me/mqtt/lib/python3.10/site-packages/teslapy/__init__.py", line 488, in api
    return self.tesla.api(name, {'vehicle_id': self['id_s']}, **kwargs)
  File "/home/me/mqtt/lib/python3.10/site-packages/teslapy/__init__.py", line 369, in api
    return self.request(endpoint['TYPE'], uri, serialize,
  File "/home/me/mqtt/lib/python3.10/site-packages/teslapy/__init__.py", line 158, in request
    response.raise_for_status()  # Raise HTTPError, if one occurred
  File "/home/me/mqtt/lib/python3.10/site-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://owner-api.teslamotors.com/api/1/vehicles/<snip>/latest_vehicle_data`

I removed the vehicle id from the api after copy/paste.

Other calls work fine, e.g. this works fine: `print(f"Original state {vehicles[0]['state']}")`

gbizeau, 2023-04-26:
You're assuming that the Telsa API (the real API this one talks to)
returned vehicle data, The better way to do this is
to iterate through the array, if it exists. If it doesn't exist, then you
need to find out why your vehicle is not being returned.

API's don't generally catch errors for you, that's your code's job.

paulhargreaves, 2023-04-26:
The API is returning vehicle data. If I examine vehicles[0] I see:

```
{
    "id": snip,
    "vehicle_id": snip,
    "vin": "snip",
    "display_name": "Car",
    "option_codes": "AD15,AF00,APFB,APH4,AU3P,BC3B,BT37,CDM0,CH15,COGB,DRRH,DV4W,FC02,FG31,FM3B,GLFR,HL32,HM30,ID3W,IL31,LTPB,MDL3,MR31,PMNG,PC30,REEU,RF3G,RS3H,S3PB,SA3P,SC04,STCP,SU3C,T3MS,TM00,TW00,UT3P,W40B,WR00,ZINV,MI02,PL30,SLR0,ST30,BG31,I36M,OSSB,AUF1,RSF1,ILF1,FGF1,CPF0,HP31,PT01,RL31",
    "color": null,
<snip>
```

paulhargreaves, 2023-04-26:
Side note: 
```
vehicles = self._tesla.vehicle_list()
print(vehicles[0].last_seen())
```

^ was a trivial example just to demonstrate the issue. My proper code does validation.

The example on [https://github.com/tdorssers/TeslaPy](https://github.com/tdorssers/TeslaPy) also does the same thing:

```
import teslapy
with teslapy.Tesla('elon@tesla.com') as tesla:
    vehicles = tesla.vehicle_list()
    print(vehicles[0]['display_name'] + ' last seen ' + vehicles[0].last_seen() +
          ' at ' + str(vehicles[0]['charge_state']['battery_level']) + '% SoC')
```

gbizeau, 2023-04-26:
OK, but in that array, is there a last_seen variable in there?

Also, the real error here is the

"404 Client Error: Not Found for url:
https://owner-api.teslamotors.com/api/1/vehicles/<snip>/latest_vehicle_data"

You have to account for the fact that the Tesla API (from Tesla) doesn't'
always return data.

I wrap all of my API calls in a try/except and expect an error.

It's not 100% guaranteed to get a return call

paulhargreaves, 2023-04-26:
There is not a last seen in there. But... why should there be? There won't be one because the car is asleep, and the doc specifically says that it's ok to use when Online: No.

Wrapping in try/expect is fine, but I've tried that call multiple times over multiple days and it always 404s, which leads me to think that a) the functionality doesn't work because Tesla changed something, so needs to be altered/removed or b) the example on the page is incomplete, and other things are needed to make it work, hence the issue raise.

tdorssers, 2023-04-26:
Tesla has indeed changed something as get_latest_vehicle_data() is no longer working. I'll start working on a fix.

paulhargreaves, 2023-04-29:
> @paulhargreaves please checkout 453df9d as it should fix this issue

Thank you. I'm away, back tomorrow so will hopefully take a look.

paulhargreaves, 2023-05-01:
@tdorssers Thank you, that works!

closed 2023-05-01 by paulhargreaves