tdorssers/TeslaPy #128: (missing_token) Missing access token parameter.

behconsci, 2023-05-21:
hey Tim, thanks for this great library!!! 
I am getting this issue below. It works well in my localhost, but on server (Debian) it is failing with this error: 

(missing_token) Missing access token parameter.

[image]

[image]

[image]

[image]

tdorssers, 2023-05-21:
This error means that oauthlib doesn't get an access token back from the Tesla server. The body is actually HTML with an access denied error instead of a JSON response.

behconsci, 2023-05-21:
yep, I see that. But do you know why it is happening? it seems TLS is old
on server? localhost is working well. Strange

fbphysik, 2023-05-27:
Hi. I have the same issue. Everything worked fine until today I set up a new virtual environment, in which I installed teslapy with all necessary dependencies.
Now I keep getting the "Missing access token parameter". 
Maybe something goes wrong with the current version of one of the packages teslapy depends on?

behconsci, 2023-05-29:
I deployed on different servers (debian, ubuntu..) and still no luck. I have TeslaPy==2.7.0 both in my localhost and on server. localhost works fine, but server doesn't. what are we missing here?

fbphysik, 2023-05-29:
I tried different versions for TeslaPy and also for the dependent packages, no change. 
Providing a refresh token obtained from a third party app didn't work either. Always the same error: MissingTokenError raised from `oauthlib/oauth2/rfc6749/parameters.py", line 451, in validate_token_parameters`.

behconsci, 2023-05-29:
what we are doing is a pure "man in the middle attack"? :D but then localhost is still working. I suspect, the scope is the issue, maybe? instead of offline_access maybe online_access .. but just a guess

fbphysik, 2023-05-29:
>  but then localhost is still working.

Lucky for you. Here nothing is working, neither server nor localhost.

behconsci, 2023-06-01:
Aaah thanks man. I will try out tomorrow morning.

HansKappert, 2023-06-06:
This saved my day.
Week
Month

Refhi, 2023-08-25:
Hey @fbphysik 

thank you so much for the tip !

had the same problem, but it was solved only by downgrading charset-nromalize and requests to the aforementioned versions, I did not have to touch the others

if that's important : I"m working in an up-to-date dockerised Domoticz where  I had : 
- requests 2.31.0 (downgraded to 2.28.1)
- certifi-2023.7.22
- charset_normalizer-3.2.0 (downgraded to 2.1.1)
- urllib3            2.0.4
- websocket-client   1.6.2
On my other computer it works with : 
- requests                           2.25.1
- certifi                            2020.6.20
- urllib3                            1.26.5
- websocket-client                   1.5.1
(charset_normalize is nowhere to be found)

If I had a guess that would be to blame a recent requests update

(thank you so much for teslapy btw :) )

tdorssers, 2023-08-26:
I believe the combination of urllib3 2.0 and Python <3.10 is causing issues. Either upgrade Python to 3.10+ or downgrade urllib3 to 1.26.x. Please let me now if this works for you.

Refhi, 2023-08-27:
I'm sorry it will be difficult for me to do more testing, though in my dockerized environment it works properly with python 3.9 and urlib 2.0.4
It broke with requests 2.31/charset_normalize 3.2, but worked again with downgrading to requests 2.28.1/charset_normalize 2.1.1 if that helps
here's my pip list (where TeslaPy is working properly) : 

Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
Package            Version 
certifi            2023.7.22
charset-normalizer 2.1.1
idna               3.4
oauthlib           3.2.2
pip                20.3.4
requests           2.28.1
requests-oauthlib  1.3.1
setuptools         52.0.0
TeslaPy            2.8.0
urllib3            2.0.4
websocket-client   1.6.2
wheel              0.34.2

Hope that'll help a bit