tdorssers/TeslaPy #131: Powerwall - cannot retrieve the mode

HeraldCoupe, 2023-05-26:
I've just upgraded to the lastest version but I am no longer able to retrieve the Powerwall mode. It looks as though the output of get_battery_data() no longer contains "default_real_mode" or "operation".

Please can you advise how I can get the mode now?

Thanks for your help.

DaveTBlake, 2023-06-09:
Yes facing the same issue here,  "default_real_mode" produces error. Would like advice on getting mode.

DaveTBlake, 2023-06-09:
While get_battery_data() no longer retrieves "default_real_mode" or  "operation" (that always seemed to be the same as "default_real_mode"), I have found from experimenting that getting data from the endpoint `api/1/energy_sites/{site_id}/site_info` does return "default_real_mode". I don't know if this is new, or if it has always done so as this is not an endpoint I have used.

To access that I added the following method to the Battery class in init.py
```
def get_site_info(self):
        """ Retrieve current site config information"""
        self.update(self.api('SITE_CONFIG')['response'])
        return self
```
Powerwall mode is then given by  `product.get_site_info()["default_real_mode"]`

HeraldCoupe, 2023-06-22:
> def get_site_info(self):
>         """ Retrieve current site config information"""
>         self.update(self.api('SITE_CONFIG')['response'])
>         return self

Thanks for this. After some fidding around I got it to work.