tdorssers/TeslaPy #135: last_seen() raises "Timestamp is in the future" error

selsrog, 2023-07-30:
Anybody else having this issue?
Played with the timezones to no eval...

rickou, 2023-09-12:
yes same here, it seems it is a rounding issue between time from servers and your local time.
I patched localy the last_seen function to round diff time to 0 if there is less than 5s in future
`    def last_seen(self):
        """ Returns vehicle last seen natural time. Raises ValueError. """
        units = ((60, 'a second'), (60, 'a minute'), (24, 'an hour'),
                 (7, 'a day'), (4.35, 'a week'), (12, 'a month'), (0, 'a year'))
        diff = time.time() - self['charge_state']['timestamp'] / 1000
        if diff < 0:
            if diff > -5:
                # round diff time to 0 if near in future to avoid false errors
                diff=0
            else:
                raise ValueError('Timestamp is in the future')
        if diff >= 1:
            for length, unit in units:
                if diff < length or not length:
                    if diff > 1.5:
                        unit = '%d %ss' % (round(diff), unit.split()[1])
                    return unit + ' ago'
                diff /= length
        return 'just now'
`

ferni7, 2023-09-24:
Just ran into the same issue. 

@rickou Can you raise a PR with this fix?

rickou, 2023-09-26:
hello there !

after some days i still have valueError exceptions.. 
strangly i parked my car 2 times at a friend's home and i get several exception raised each time, 
(Don't know why specifically at this place !)
So this doesn't seem to be sufficient to compensate all tesla server/car and local computer time differences...
i added some trace at my home to seen how much this can drift..

but after some thinking, for me there is 2 solutions:
1) If exception need to be kept in library, at user side, add a try/except like this:
          lastseen="just now"
          try:
            lastseen=veh.last_seen()
          except ValueError as e:
            pass

2) Simply remove the exception in last_seen function..
Nohting to do more, and in case of diff in future we get "just now", which is quite good in fact..

commit ae596b7a (2023-10-07): Removed exception to fix #135
closed 2023-10-07 by tdorssers