tdorssers/TeslaPy #136: No longer able to set Powerwall to self_consumption mode

HeraldCoupe, 2023-07-31:
I am no longer able to set the Powerwall to 'self_consumption' mode. I'm still able to set it to 'autonomous' mode.
I'm guessing something has changed at Tesla's end. 

Anyone got any ideas?

rob0101, 2023-08-17:
I'm new to teslaPy and not fluent in python.

I've had no luck getting a simple script to toggle between the two states. I was able to change the minimum reserve, so it may be I'm hitting a similar issue to HeraldCoupe.

Can someone share a simple code sample that changes the state as a starting point.

Thanks

HeraldCoupe, 2023-08-17:
Thank you Rob0101. From my limited testing it seems that in addition, the backup reserve also needs to be different to that currently set otherwise it doesn't change modes.

DaveTBlake, 2023-10-06:
Interesting experiences guys, I've not been setting mode since last winter (switching to autonomous to get the faster charge rate over the cheap electricity supply period that I am tricking the Powerwall into charging from the grid by changing the reserve percentage). I always set both reserve percentage and mode so maybe I would not notice this issue.

I wonder if this is another aspect of the Tesla API randomly deprecating the `powerwalls` endpoints in favour of  `energy_sites`. The endpoints I use are:
```
  "OPERATION_MODE": {
    "TYPE": "POST",
    "URI": "api/1/energy_sites/{site_id}/operation",
    "AUTH": true
  },
  "BACKUP_RESERVE": {
    "TYPE": "POST",
    "URI": "api/1/energy_sites/{site_id}/backup",
    "AUTH": true
  },
```
Note I'm using `energy_sites` for both, and first I get `api/1/energy_sites/{site_id}/site_info` ("SITE_CONFIG" in endpoints.json) using a routine I added to the battery class (which maybe populates the {site_id} ?) .

@rob0101 in your example you use a mix of  `energy_sites` and `powerwalls` since 
```
"BATTERY_BACKUP_RESERVE": {
    "TYPE": "POST",
    "URI": "api/1/powerwalls/{battery_id}/backup",
    "AUTH": true
  },
 ```` 
 Maybe that is significant? Of course it could all change tomorrow. Hope this could be of use to either of you.

tdorssers, 2023-10-07:
Commit https://github.com/tdorssers/TeslaPy/commit/50bdbfba815e113a5982690e7124b80aa8d1da2d removed the depricated powerwall API. Please use the energy site API.

rob0101, 2023-10-08:
I've updated to the new teslapy code and my revised script (below) is working.

import sys
new_mode = ""
new_MRL = ""
if len(sys.argv)>1 : new_mode = sys.argv[1]
if len(sys.argv)>2 : new_MRL = int(sys.argv[2])

import teslapy
`from teslapy import Tesla, Battery
`with teslapy.Tesla('elon@tesla.com') as tesla:
`       batteries = tesla.battery_list()
`       battery = batteries[0]
`       if new_MRL != '':
`            battery.set_backup_reserve_percent(new_MRL)
`       if new_mode != '':
`            battery.set_operation(new_mode)
`
`      battery.get_site_info()
`      print("Mode:" + battery["default_real_mode"])
`      print("Reserve:" + str(battery["backup_reserve_percent"]))