tdorssers/TeslaPy #143: Solar panel power only updates sporadically

guillochon, 2023-09-22:
Not sure if this is just a change in the data that Tesla reports via their API, but I've been finding the past several weeks that the amount of power Tesla reports via their API for solar panels only updates very sporadically, like once or twice a day now (used to be every 15 minutes or so). I'm using `solar_list()` to retrieve the panels and then looking at the `solar_power` value there.

Perhaps there might be another API endpoint for retrieving this data more reliably? As it stands the API request is much less useful given the very infrequent updating.

tdorssers, 2023-10-07:
Please checkout commit 50bdbfb and try to use get_site_info() or get_site_data().

guillochon, 2023-10-08:
I verify that this is working for me now; however the fix is not actually related to this commit. It looks like Tesla used to provide an up-to-date `solar_power` even if you didn't call `get_site_data()`, that no longer appears to be the case and you need to explicitly call that function now to get up to date numbers. Unrelated I think to the change you made.

closed 2023-10-08 by guillochon