tdorssers/TeslaPy #145: 410 error on get_battery_status

doubledrat, 2023-10-06:
Get battery status Started failing last night with the following error -

requests.exceptions.HTTPError: 410 Client Error: https://powergate.prd.sn.tesla.services:443/api/powerwalls/idremoved/fullstatus => Gone for url: https://owner-api.teslamotors.com/api/1/powerwalls/myidremoved

yanschneider, 2023-10-06:
For me as well. It seems that the teslamotors api is no longer available under the given URL.

DaveTBlake, 2023-10-06:
Same here, my script using get_battery_data failed last night. Looks like the endpoints starting `api/1/powerwalls/` no longer work :(

However I have had success getting the Powerwall data I want by swapping to use  `api/1/energy_sites/{site_id}/site_info`, followed by `api/1/energy_sites/{site_id}/live_status`, calling the first is necessary for the second API call to return fully populate data. These endpoints are in endpoints.json as "SITE_CONFIG" and "SITE_DATA" respectively, and to access them I have added routines to the battery class in `_init_.py`  as follows
```
    def get_site_info(self):
        """ Retrieve current site generation data """
        self.update(self.api('SITE_CONFIG')['response'])
        return self

    def get_site_data(self):
        """ Retrieve current site generation data """
        self.update(self.api('SITE_DATA')['response'])
        return self
```

JuhaSeppa, 2023-10-06:
I've been fetching Tesla Powerwall power timeseries data, which stopped working via the `api/1/powerwalls/`. I was able to retrieve the same data from the 'CALENDAR_HISTORY_DATA' api endpoint with `kind='power'` and `date_end='time in isoformat'` arguments to get the data I wanted.

mquilling0, 2023-10-06:
I am having the same issue but for setting operation mode - any workaround for this?
```
battery.set_operation("self_consumption")

410 Client Error: https://powergate.prd.sn.tesla.services:443/api/powerwalls/{xxxx}/operation => Gone for url: https://owner-api.teslamotors.com/api/1/powerwalls/{yyyy}/operation
```

j4mie, 2023-10-06:
@mquilling0 it looks like this works:

```py
battery.api("OPERATION_MODE", default_real_mode="self_consumption")
```

There seem to be two endpoints for changing operation mode. `set_operation` uses the one named `BATTERY_OPERATION_MODE` in the `endpoints.json` file, which points at `api/1/powerwalls/{battery_id}/operation`. This is the one that no longer works. `OPERATION_MODE`, on the other hand, uses `api/1/energy_sites/{site_id}/operation`, which does still work. Presumably `set_operation` needs to be changed to use this endpoint.

JuhaSeppa, 2023-10-06:
> I am having the same issue but for setting operation mode - any workaround for this?
>
> ```
> battery.set_operation("self_consumption")
>
> 410 Client Error: [...]

Depending on what you want to achieve, you might be able to realise the desired functionality by keeping the operation mode unchanged and just manipulating the backup reserve percentage.
I’ve implemented charge (backup = 100%), standby (backup = current charge level from the local API) and self consumption (backup = whatever percentage you want to keep in reserve for a power cut) this way while keeping mode as autonomous. Overnight cheap electricity is configured to the Powerwall and I use these operations only when I want to achieve something else.

mquilling0, 2023-10-06:
thank you @j4mie that worked!

mquilling0, 2023-10-06:
@JuhaSeppa i have a very specific need that neither mode can achieve alone

export from sun up to 9am
charge from 9am to 4pm
export from 4pm to 8pm but dont drop below 65% (we rarely use the grid but this is the only time we might - which gives us the ability to be awake and limit usage during an outage)
use battery overnight (but turn off any appliance if % drops below 10% - rarely happens)

i can achieve this exact profile and rarely use any grid energy + maximize my energy export credit. our yearly bill is negative.

using time based control it does all kinds of dumb stuff and pulls a few kW/day from the grid.

HeraldCoupe, 2023-10-07:
I presume at some point TeslaPy will be updated to reflect the change that Tesla have implemented.
Any ideas as to when?
Thanks to the developers, TeslaPy is a life-saver!

yanschneider, 2023-10-07:
`battery.api("OPERATION_MODE", default_real_mode="self_consumption")`

still gives me the 410 error unfortunately.

```
tesla = teslapy.Tesla(email_address)
battery_list = tesla.battery_list()
battery = battery_list[0]
battery.api("OPERATION_MODE", default_real_mode="self_consumption")
#battery.set_operation('self_consumption')
battery.set_backup_reserve_percent(10)
```

tdorssers, 2023-10-07:
I don't own a powerwall, so I cannot test it, but commit 50bdbfb should fix this, based on @DaveTBlake's code.

mquilling0, 2023-10-07:
@tdorssers tested and works, thx!

HeraldCoupe, 2023-10-08:
> I don't own a powerwall, so I cannot test it, but commit [50bdbfb](https://github.com/tdorssers/TeslaPy/commit/50bdbfba815e113a5982690e7124b80aa8d1da2d) should fix this, based on @DaveTBlake's code.

Thank you so much, I've also tested it and got it working.

jasonacox, 2023-10-08:
@tdorssers Thank you! This does indeed fix the issue. The schema changed a bit for `battery` so anyone using this to read battery data will need to a adjust ([example](https://github.com/jasonacox/pypowerwall/commit/d9a8278624f46c2655dcd6028d8a766898f50c43)).

Test:

```bash
# Pull latest TeslaPy - use latest until TeslaPy releases a new PyPI package
pip install -e git+https://github.com/tdorssers/TeslaPy.git#egg=teslapy

# Pull latest set-reserve.py 
wget https://raw.githubusercontent.com/jasonacox/pypowerwall/main/tools/set-reserve.py

# Test
python3 set-reserve.py --read
READ: Current Battery Reserve Setting: 20% for 2 Powerwalls
```

DaveTBlake, 2023-10-09:
>using time based control it does all kinds of dumb stuff and pulls a few kW/day from the grid.

@mquilling0 doesn't it just!!!

>I don't own a powerwall, so I cannot test it, but commit https://github.com/tdorssers/TeslaPy/commit/50bdbfba815e113a5982690e7124b80aa8d1da2d should fix this, based on @DaveTBlake's code.

Thanks @tdorssers , sorry I didn't get my act together and raise a PR just left code suggestions. I'm very grateful for everything you have done making the API avialable, my Powerwall would be unmanageable without it.