tdorssers/TeslaPy #147: latitude and longitude are not in drive_state

HansKappert, 2023-11-01:
I ran 
`python3 menu.py`

After logging in, this error is printed at the console:
`File "/Users/Hans/Downloads/TeslaPy-master/menu.py", line 37, in show_vehicle_data
    coords = '%s, %s' % (dr['latitude'], dr['longitude'])
KeyError: 'latitude'`

heraldb, 2023-11-10:
The is already a patch for the Elixer library: https://github.com/adriankumpf/teslamate/pull/3373/commits/5058a6fa07f3fae2f1552c7150b28481446967c5

(so in fact you need to add "?endpoints=location_data" to the url)

RichieB2B, 2023-11-11:
While the quick fix of https://github.com/tdorssers/TeslaPy/commit/38ff7da6e5a7aa1fcd5d39b84aa1d89c9cff50b5 probably works, it will show the location icon in the car whenever TeslaPy requests VEHICLE_DATA. IMHO it would be better to create a separate endpoint to specifically retrieve location data.

robharman, 2023-12-18:
> While the quick fix of [38ff7da](https://github.com/tdorssers/TeslaPy/commit/38ff7da6e5a7aa1fcd5d39b84aa1d89c9cff50b5) probably works, it will show the location icon in the car whenever TeslaPy requests VEHICLE_DATA. IMHO it would be better to create a separate endpoint to specifically retrieve location data.

Submitted https://github.com/tdorssers/TeslaPy/pull/153 to implement this suggestion, as well as adding another endpoint/method to get only basic vehicle data without location and config, or the other endpoints as I often find myself needing only the basics. If there's enough need for it, happy to create something more extensible that would allow selective additional endpoints to be queried through another method. 

Though, maybe updating the default `get_vehicle_data()` may be a better idea. Set it to default to all endpoints, but if options are requested only query basic data and those endpoints perhaps?